BruceDevices/firmware #314: Custom GIF viewer in startup

Ademolght, 2024-10-06:
**Is your feature request related to a problem? Please describe.**
Custom GIF viewer for bruce firmware using jpg frame i think can be done esealy using the jpg view
 and i also found this repo for GIF viewer with M5stack https://github.com/tobozo/ESP32-GifPlayer

eadmaster, 2024-10-06:
it should require this library only 
https://github.com/bitbank2/AnimatedGIF/blob/master/examples/TFT_eSPI_memory/TFT_eSPI_memory.ino

Flipper0 uses a custom file format for animations, but it is B&W only, so i think it's better to stick with standard GIFs:
https://github.com/UberGuidoZ/Flipper/tree/main/Graphics/Animations/PYX

Ademolght, 2024-10-10:
@eadmaster so i close it

closed 2024-10-10 by Ademolght

eadmaster, 2024-10-10:
Gif images could also be used as replacement for the current hardcoded menu icons, i'm testing them here:

https://github.com/eadmaster/Bruce/commit/c7e831e654429eb002030f20bff8f87d3109dc09