BruceDevices/firmware #315: RFID write dump fonction

LEGOATDU49, 2024-10-06:
On the Bruce firmware, I tried to copy my badge and put the data on a blank block 0 rewritable badge, when I cloned the uid it failed and when I wrote the data it gave me success but the badge was not good. Then, I tried the Mifare classic tool application for Android equipped with an NFC reader and using the "WriteDUMP" function and it worked thanks to this tutorial without changing the uid:https://www.youtube.com /watch?v=tLtF5b-Hww0

Can you do something for Bruce?

IncursioHack, 2024-10-24:
> On the Bruce firmware, I tried to copy my badge and put the data on a blank block 0 rewritable badge, when I cloned the uid it failed and when I wrote the data it gave me success but the badge was [...]

Could you test again with the BETA version? @rennancockles  made several changes to NFC/RFID that may make sense to you.

IncursioHack, 2024-11-02:
@LEGOATDU49 I need your feedback on testing the latest versions (BETA), please attach the Serial Log.

bmorcelli, 2024-12-07:
no answer :(

closed 2024-12-07 by bmorcelli

pacaja, 2024-12-12:
I have the same problem. I use these tags from Aliexpress:
https://nl.aliexpress.com/item/1005005497385298.html?spm=a2g0o.order_list.order_list_main.21.17ba79d296OTTs&gatewayAdapt=glo2nld

I cannot write te UID to the tag. If I clone an UID bruce is saying it was succesfull but the UID is not changed,  If I use Mifare Classic Tool on Android it is working. The app uses a key (FFFFFFFFFFFF) to write to block 0.

I have installed the latest BETA of Bruce on my Lilygo T-Embed CC1101.