BruceDevices/firmware #316: BLE Hacking Ideas?

Jethro-San, 2024-10-06:
would it at all be possible to make a program that forces a device to pair with the cardputer so that you can control it with the media cmds?

IncursioHack, 2024-10-08:
Forcing a connection is extremely complex, I haven't seen anything like that running this, if it exists it would probably be a CVE. What I think is possible is to change the MAC to that of a device the victim trusts and try to connect by exploiting that trust.

Jethro-San, 2024-10-09:
How would you find out what MACs are trusted by the target?

7h30th3r0n3, 2024-10-09:
> Forcing a connection is extremely complex, I haven't seen anything like that running this, if it exists it would probably be a CVE. What I think is possible is to change the MAC to that of a device the victim trusts and try to connect by exploiting that trust.

sounds like CVE-2023-45866 
https://github.com/pentestfunctions/BlueDucky
😘

cinouw, 2024-10-13:
> 
> Forcing a connection is extremely complex, I haven't seen anything like that running this, if it exists it would probably be a CVE. What I think is possible is to change the MAC to that of a device the victim trusts and try to connect by exploiting that trust.

I don't think forcing a connection is possible with the esp32 hardware instead you can make it so it spams pairing requests to targeted device until the user accepts the connection and you can make the name and the device type that appears when the pair request is sent changeable so the we can make it covincing the the only "force pair" thing that i think is possible using esp32 hardware

DMac0117, 2024-10-28:
Using BLE to recreate the audible ping feature from an Apple IWatch to an Apple Iphone?

closed 2025-05-24 by bmorcelli

Huzzla101, 2025-05-24:
I have one, i see you  just updated BLE with more candy but we need one way
to get connected to the "target" to, not just spamming,  think if we could
manage to actually connect to device if they pushed accept when spamming ?
i know its impossible at first but lets slowly find a way ... All people
with Bluetooth tech information  COME SHARE WITH  Mr  Pirata all you know
..

One way is to make a NRF24 SNIFFER/AUTENTICATOR that READS all nrf showen
at spectrum but read it in raw format, then its possible to find ANY
electric device that sends signals , even your mouse  and all bluetooth
stuff etc..... To keep it short, nrf24 sniffer can read anything at 2,4ghz
zAND one with  little experience quickly will find any device and  also it
sniffs the receiver of device to and its not hard to store any signals just
like IR just that we have the whole bluetooth and 2,4 ghz spectrum.. We
already there with our spectrum analyzer we just need to re arrange the
info we see at the strings popping up over to raw format text and
digits....  That is the LAST part of the NRF24 to keep it complete.... Only
Flipper can do this for now , no other firmware ....

FAMOUS things like mousejacker is not possible without first sniffing the
usb dongle ID and that is sent over the NRF24  ..... Then you can actually
sniff the device ID and when you have that (the usb dongle ) then you can
send nrf24 signals off course that you recorded to the mouse such as
movement OR send a bad usb ... you get my point ? Parts of Europe have
already started taking over the electricity as they to send nrf24 signals
and people have been studying them enough to be able to turn it on off and
alot of more options... MAJOR operations for the future .-

So i advice that we slowly when we have the possibility starts to make a
nrf24 Sniffer