BruceDevices/firmware #323: Rf bug

cinouw, 2024-10-08:
Hi bruce team

I recently downloaded 1.6.1 firmware to test the frequency detection feature the feature works great but it has a bug

The way the bug happens is when I press next to set the frequency range it dost start to detect the frequency but after that when I try to read the data keeps getting stuck in the frequency detection mode and I can't read any data without restarting the cardputer

IncursioHack, 2024-10-24:
> Hi bruce team
> 
> I recently downloaded 1.6.1 firmware to test the frequency detection feature the feature works great but it has a bug
> 
> The way the bug happens is when I press next to set the frequency range it dose start to detect the frequency but after that when I try to read the data keeps getting stuck in the frequency detection mode and I can't read any data without restarting the cardputer

Please perform a new test on the BETA version. 

Furthermore, keep your device connected to the computer via Serial and collect the Logs. Another point is to inform which device and which RF you are using. This makes it easier for us to reproduce the scenario.

If possible, record a short video about the problem and attach it to this issue.

cinouw, 2024-10-24:
@IncursioHack hi i sorry for the late reply i will retry with beta build

While testing i was using cc1101

IncursioHack, 2024-10-26:
@cinouw 
I need some information to better understand what is happening:
1 - Is the device Cardputer, StickC, CYD...?
2 - Could you carry out the test on the most current version? 1.6.2
3 - Did you select the CC1101 in RF > Config > Module > CC1101?
4 - Have you checked the frequency of the device you are testing? (315mhz, 433.92mhz....)
5 - Could you provide the Serial LOGs together?
6 - Would you have a video to demonstrate what is happening?

cinouw, 2024-10-26:
@IncursioHack Hi

1- I'm using cardputer 
2- the happend on ver. 1.6.1 I didn't try it with 1.6.2 after I try it with 1.6.2 I will inform you 
3- yes I selected cc1101
4- i know the device frequency but I wanted to try the frequency detection feature 
5- I will provide you with the log's after i install 1.6.2
6- ofc I will take a video if the same bug happened in 1.6.2

cinouw, 2024-11-05:
@IncursioHack 

Hi so I recorded this video showcasing the bug 
Note : sorry for my bad english I am still learning

https://drive.google.com/file/d/18JZ702BCXgAoh5e_dWsR7b0peMPeVpmx/view?usp=drivesdk

bmorcelli, 2024-11-05:
Let me help you with a fix, without loosing your bruce.conf (and wifi passwords and stuff saved there)

* RF>Config>Rf Frequency > Type "433.92"
Or 
* RF>Scan/Copy>(Left arrow) Range> choose "387-464 MHz" and you can keep trying receiving the right frequency..

Try not send the signal too close from the receiver, because it is catching the 5 frequencies with good RSSI, and getting the best value of them... so if you are too close, you will have high strength signal on the raise of the "normal curve", and because we are using a wide bandwidth filter here https://github.com/pr3y/Bruce/blob/3415c848c0dbd2aad853925951261b8505d4409d/src/modules/rf/rf.cpp#L499 you probably will catch the wrong frequency..

Try sending the signal from 1,5m away for testing.

cinouw, 2024-11-05:
@IncursioHack 

Hi thanks for the response 

I will retry with the parameters that you mentioned and I will update you

bmorcelli, 2024-12-07:
fixed on beta

closed 2024-12-07 by bmorcelli