BruceDevices/firmware #326: Main menu customization feature (hide/show items)

eadmaster, 2024-10-10:
Add a config option to hide/show main menu items according to user preference (saved in `bruce.conf`).

In my case i have these items i'd like to hide, since i don't have the modules required to use them: RFID, FM, NRF24.

This should help with the increasing number of  items in the main menu making the navigation too slow.

It could also justify moving some items from the "Others" menu, like BadUSB  as suggested here: https://github.com/pr3y/Bruce/issues/325  https://github.com/pr3y/Bruce/issues/123

eadmaster, 2024-11-30:
Initial implementation added in https://github.com/eadmaster/Bruce/commit/81bb8fc5f6b9695c3278cc3e07513d9833f3eaa4

Possible future enhancements:

 - add a menu in "Config" to interactively choose the menus to disable
 - move the disabled menus inside "Others"
 - add another array in `bruce.conf` to customize the menu items order (and implicitly disable them)

mxrxtr, 2025-02-24:
Is this now implemented?,if yes how do I hide things

eadmaster, 2025-02-24:
Manually edit 'bruce.conf'.

See the example in
https://github.com/pr3y/Bruce/pull/500

bmorcelli, 2025-06-19:
Added on Beta (18/06/25)

closed 2025-07-22 by eadmaster