BruceDevices/firmware #333: Need some information

crxyox, 2024-10-10:
I was looking for any documentation on what other devices and modules are needed for features like badusb NFC and RFID (I have an m5stickc+)

IncursioHack, 2024-10-10:
https://discord.com/channels/1264994142769909854/1276567989654782002/1276569089816395848

IncursioHack, 2024-10-24:
Essential Pack (Tested on Bruce):

- Cardputer: https://s.click.aliexpress.com/e/_DDUzXPN
- SD Card Sniffer (Important): https://s.click.aliexpress.com/e/_DEeiBz1
- M5StickC Plus 2: https://s.click.aliexpress.com/e/_DlnmePt
- CYD-2432S028 (Cheap Yellow Display): https://s.click.aliexpress.com/e/_DezqszL
- MicroSD Module (Stick): https://s.click.aliexpress.com/e/_mtxmLeo
- IR Remote (IR): https://s.click.aliexpress.com/e/_DePIKer
- GPS (Wardriving): https://s.click.aliexpress.com/e/_DkF8JPd
- CC1101 (RF - SubGhz): https://s.click.aliexpress.com/e/_DlaeenD
- Chameleon Ultra (RFID/NFC/Emulate): https://s.click.aliexpress.com/e/_mN6vZiG
- NRF24L01-PA-LNA (Jammer BLE/Wifi 2.4Ghz): https://s.click.aliexpress.com/e/_DcRrayF
- Transmissor FM SI4713 (Radio FM): https://s.click.aliexpress.com/e/_DCKYlfz
- Cable Dupont (For all): https://s.click.aliexpress.com/e/_DBMjN2B
- Cable Grove2Dupont (For all): https://s.click.aliexpress.com/e/_DDrQLjv
- CH9329 Mouse/Keyboard HID (Stick): https://s.click.aliexpress.com/e/_DmzsZih

closed 2024-10-24 by IncursioHack