BruceDevices/firmware #334: NRF24 and RC522 modules not being detected

RarefiedMars415, 2024-10-11:
My NRF and RFID modules cant get detected by Bruce? for the NRF24 im not sure but i do know that the rc522 is working since the module is lighting up but still saying that the rfid module is not found same goes for the NRF

cinouw, 2024-10-13:
For nrf24 the make sure it is receiving the correct 3.3v and make sure the connections to the SD card pins and the grove port pins are correct and don't use long wires because the nrf is sensitive to long wires 

For first make sure it is receiving 3.3v if it is the try flipping sda and scl pins

TesseractXT, 2024-10-19:
Just press and hold the power button to turn it off. Now again turn it on. Now go to config. It'll be detected for sure.

IncursioHack, 2024-10-24:
> Meus módulos NRF e RFID não podem ser detectados por Bruce? Para o NRF24 não tenho certeza, mas sei que o rc522 está funcionando, pois o módulo está acendendo, mas ainda diz que o módulo RFID não foi encontrado, o mesmo vale para o NRF

Dear all, please review the connections of the devices mentioned, we received several reports of various devices not working which we found to be just loose cables, poorly connected or connected in the wrong way. Additionally, we suggest you test the beta version of bruce at https://bruce.computer. Enjoy and review our wiki https://github.com/pr3y/Bruce/wiki

closed 2024-10-24 by IncursioHack