BruceDevices/firmware #343: I need connection scheme

Sonys9, 2024-10-15:
can you give me a scheme for RC-522 to m5stickc plus2? (not via discord, discord is banned in my country)

Sonys9, 2024-10-15:
[image]

rennancockles, 2024-10-15:
[image]

Use I2C connection (SDA SCL GND 3.3V) and connect to grove port like this

[image]

rennancockles, 2024-10-15:
[image]

Use the I2C connection (SDA SCL GND VCC)

rennancockles, 2024-10-15:
Be careful that RC522 uses 3.3V and grove port outputs 5V. You need to use the 3.3V from stickc plus2 gpio

Sonys9, 2024-10-18:
[image]

Sonys9, 2024-10-18:
[image]

closed 2024-10-24 by IncursioHack