BruceDevices/firmware #345: T-WATCH ESP32-S3,Could it be supported by Bruce ?

YANGKUN87, 2024-10-17:
[image]
This watch is based on ESP32-S3 and is quite portable. Can we add support for it?

bmorcelli, 2024-10-18:
Interesting device!

Maybe if Lilygo sent me one of these I port Launcher and Bruce to it 😜

YANGKUN87, 2024-10-20:
> Interesting device!
> 
> Maybe if Lilygo sent me one of these I port Launcher and Bruce to it 😜

This programming watch is not expensive, you can buy one to play with and wait for you to port it. thank you！

rennancockles, 2024-10-20:
@YANGKUN87 if it's not expensive you could buy one and send it to Pirata so he can work on the port then.

Jaynehy, 2024-10-24:
> Interesting device!
> 
> Maybe if Lilygo sent me one of these I port Launcher and Bruce to it 😜

try to contact them maybe you will get one

Asmodeux2, 2024-10-26:
<!DOCTYPE html>
<html lang="en">
<head>
    <meta charset="UTF-8">
    <meta name="viewport" content="width=device-width, initial-scale=1.0">
</head>
<body>
    <h1>🌟 Proposal: Use LILYGO T-Embed ESP32-S3 for Bruce Firmware 🌟</h1>
    [image]
    <p>
        Hello team! 👋
    </p>
    <p>
        I would like to propose the use of the LILYGO T-Embed ESP32-S3 module as an alternative to the Flipper Zero for the Bruce firmware. Here are the reasons why this could be a beneficial move:
    </p>
    <ul>
        <li>💡 <strong>Enhanced Performance:</strong> The ESP32-S3 offers improved processing power and memory, which can enhance the overall performance of the Bruce firmware.</li>
        <li>🔒 <strong>Security Features:</strong> The module comes with advanced security features that can be leveraged to ensure the safety and integrity of the firmware.</li>
        <li>🌐 <strong>Wide Compatibility:</strong> The ESP32-S3 is widely supported and has a large community, making it easier to find resources and support for development.</li>
        <li>💰 <strong>Cost-Effective:</strong> The LILYGO T-Embed ESP32-S3 is a cost-effective solution, making it a more affordable option for our project.</li>
    </ul>
    <p>
        By adopting the LILYGO T-Embed ESP32-S3, we can ensure that our project remains at the forefront of technology while also being cost-effective and secure. 🚀
    </p>
    <p>
        Looking forward to your thoughts and feedback! 🤔
    </p>
    <p>
        Best regards,<br>
        Asmodeux 🤝
    </p>
</body>
</html>

bmorcelli, 2024-10-26:
Good news guys!!

@YANGKUN87 Lilygo sent me this T-Watch S3, and 2 more devices!! So probably in a near future we will have bruce running on this guy

@Asmodeux2 T-Embed is being ported.. I'm having problems with the CC1101 and PN532 integration... but it has been implemented in my fork, need some adjustments..

Asmodeux2, 2024-10-26:
Man, I was just making a suggestion. Basically, thank you for the project.

bmorcelli, 2024-10-26:
yes, I know and it's ok.. no effence taken! lol

sacriphanius, 2025-02-28:
Me and my two friends from the same team are looking forward to it :)

closed 2025-05-24 by bmorcelli