BruceDevices/firmware #347: implementation of 5G features.

Lordex-x, 2024-10-17:
Is it possible to somehow adapt a cardputer to reach 5G Wi-Fi frequencies? If it is possible, how can it be done? Could this feature be implemented in the Bruce project in the future?

cinouw, 2024-10-18:
The esp32 s3 used in m5 products only supports 2.4ghz wifi I don't think 5ghz is possible with m5 hardware

IncursioHack, 2024-10-24:
The ESP32's processing is extremely small, I don't know if you've already carried out any file transfer tests, but it doesn't reach 1mbps, I don't think there would be any reason to implement 5G in the ESP32, other than the manufacturing costs. For this, my suggestion might be to use a Raspberry Pi with a Kali and carry out attacks with aircrack and other tools. If we compare, not even Flipper presented a solution to connect to 5Ghz networks, even using an RP2040. There are rumors about an ESP32-C5 that would supposedly have dual band (2.4 and 5Ghz), but as far as I know it hasn't been manufactured yet, so I don't think we'll have anything like that soon.

closed 2024-10-24 by IncursioHack