BruceDevices/firmware #357: Reading arduino serial output on m5stickC plus 2?

bisaw37, 2024-10-25:
I don't know if this is already possible or not, but I think it would be very useful to me and many others if you could read serial output from an arduino through USB the same way it is read in the "serial monitor" in arduino IDE. Just a little bit ago I was trying to get a diy tachometer to work for my go kart, and I think it would be great to be able to get my pocket sized device and easily diagnose any bugs in my code. Out of all my feature requests on here, this one is by far the one I think would help the most to anyone, not just me.

bisaw37, 2024-10-25:
Is there a module that can be connected to give USB host capabilities to a board without it? I have the arduino uno, which does not have USB host.

eadmaster, 2024-10-26:
It's the other way around: you may need a module to add usb host on the esp32 board, like this one:
https://shop.m5stack.com/products/usb-module

Since this requires SPI bus usage, i'd rather give it a try with esp32s3 boards like the Cardputer that would only require the groove port for connection:
https://github.com/hathach/tinyusb

bisaw37, 2024-10-26:
Could a c9329 module like this work or do I just not really get what you are saying?
https://a.aliexpress.com/_mOHWTfe

bisaw37, 2024-10-26:
Ok then, I feel like I got it, thanks.
So basically, the verdict is, it's not possible (or just not practically feasible) to be able to see serial output on the plus2?

eadmaster, 2024-10-26:
Direct uart pins connection with a level shifter should be the easier way in this case:
https://github.com/pr3y/Bruce/issues/357#issuecomment-2437087057

bisaw37, 2024-10-26:
Ok then, I'll look into a uart connection. Thanks.

closed 2024-10-26 by bisaw37

eadmaster, 2024-10-26:
you may also try with a voltage divider to get the same effect

https://www.reddit.com/r/arduino/comments/sr9qlx/i_tried_to_make_a_voltage_divider_for_connecting/
https://www.reddit.com/r/arduino/comments/1ec9h2c/level_shifting_for_serial_communication/