BruceDevices/firmware #358: CYD2USB CC1101 support

Grzybekrk, 2024-10-25:
**Describe the bug:**
<i have a special slot for cards like for cardputer and instead of g1 and g2 i used pins 21 and 22 

All looks conected correctly good voltage on module but its not working any idea why [image]

I tried both slots this one below also and still not working 

In settings i see. Tx and rx so maybe I need use them ? 

Pls help

bmorcelli, 2024-10-25:
CC1101 pins are hardcoded, and will work only after this last update (1.6.2)... The connector must go to the one beside the SDCard (gpio 22 and 27)

Then you must set the CC1101 in: RF>Config>Module>CC1101 on SPI, if it gives you an error, try switching the pins that come grom 22 and 27 

Explanation: RfTx pin and RfRx pin are used only for other modules that work with a single pin as FS1000A, if ots shown on the RF submenu, cc1101 isn't set

Grzybekrk, 2024-10-25:
@bmorcelli you are a king ;) it works now I will test it any issues I will let you know

10Q much appreciate

bmorcelli, 2024-10-25:
Thats great!! So I will close this issue!!

closed 2024-10-25 by bmorcelli

AwayGloom, 2024-10-29:
Mine sometimes connects to the module and sometimes doesn't, but when I click on scan the esp32 freezes