BruceDevices/firmware #359: RC522 connection diagram

Teapot321, 2024-10-25:
[image]

Grzybekrk, 2024-10-25:
Is this working on CYD2USB??

Grzybekrk, 2024-10-25:
[image]

Also is there any change to use this one rc522

IncursioHack, 2024-10-25:
https://docs.m5stack.com/en/unit/rfid2

M5Stack: RFID2 compared to the previous generation RFID UNIT only makes the replacement of read and write IC (RC522->WS1850S), there is no difference in function.

Apparently it should work with CYD, as long as it is connected correctly. In this case, the module selection option would probably be M5 RFID2.

Grzybekrk, 2024-10-26:
Hi 
I have tried and cannot figure out how to run it. I have conected SDA to pin 27 (also  chceked 22) SCK to CLK, MOSI to CMD, MISO to DAT0 and in config testing all 3 options for module but its not working are you able help [image]

rennancockles, 2024-10-27:
Miso, miso and sck are SPI pins. SDA is an I2C pin. Also, it won't work with SPI, you have to connect the module using I2C for this to work. I don't have this module, so I cannot help much. You can google RC522 I2C connection though

rennancockles, 2024-10-27:
Also, the CYD board is still under development. I don't know the pins are configured in the code for this board. Do you have any other board to test ?

Grzybekrk, 2024-11-06:
Hey i Just tested different rifd and works but some functions desnt do you know if we can clone tags [image]

rennancockles, 2024-11-06:
It depends on your tag. You need a special tag to be able to write into block 0 to clone. Check out the wiki https://github.com/pr3y/Bruce/wiki/RFID

bmorcelli, 2024-12-07:
We found out last week that RC522 doesn't work on I2C without modding (cut 2 trails, one of them below the chip)..

we are studying to implement UART communication (what need to cut ine trail in the board)

for refference: https://forum.arduino.cc/t/rc522-rfid-rc522-switching-spi-to-uart-interface-or-i2c-possible/425741

rouing, 2025-01-01:
Stale, backburner.

closed 2025-01-01 by bmorcelli