BruceDevices/firmware #361: Feature request: Router Keygen for Default WPA Passwords

BrainRape23, 2024-10-26:
hi i have a feature request

any chance of implementing the yolosec router keygen

it calculates the default wpa 2 key based on the mac its quite a usefull tool

the following brands are supported

    Thomson based routers ( this includes Thomson, SpeedTouch, Orange, Infinitum, BBox, DMax, BigPond, O2Wireless, Otenet, Cyta , TN_private, Blink )
    DLink ( only some models )
    Pirelli Discus
    Eircom
    Verizon FiOS ( only some routers supported)
    Alice AGPF
    FASTWEB Pirelli and Telsey
    Huawei (some InfinitumXXXX)
    Wlan_XXXX or Jazztel_XXXX
    Wlan_XX ( only some are supported)
    Ono ( P1XXXXXX0000X )
    WlanXXXXXX, YacomXXXXXX and WifiXXXXXX
    Sky V1 routers
    Clubinternet.box v1 and v2 ( TECOM-AH4XXXX )
    InfostradaWifi
    CONN-X
    Megared
    EasyBox, Arcor and Vodafone
    PBS (Austria)
    MAXCOM
    PTV
    TeleTu/Tele2
    Axtel, Axtel-xtremo
    Intercable
    OTE
    Cabovisao Sagem
    Alice in Germany
    Speedport
    UPC (UPCXXXXXXX)

https://github.com/yolosec

thx

eadmaster, 2024-10-26:
Related/duplicate of https://github.com/pr3y/Bruce/issues/131

Could be implemeted as a variation of this script:
https://github.com/pr3y/Bruce/blob/main/sd_files/interpreter/wifi_brute.js

BrainRape23, 2024-10-26:
i will look into it maybe with the help gpt i can get it to work

IncursioHack, 2024-10-26:
Hi, in view of @eadmaster 's comment about duplication and the possibility of execution via JS Interpreter, I will be closing this issue. Thank you for everyone's support!

closed 2024-10-26 by IncursioHack