BruceDevices/firmware #362: Add RAW support with cc1101

Teapot321, 2024-10-26:
Cathack has had this for a long time, but Bruce (the best firmware) does not

Petitoursbr1, 2024-10-26:
Yeah I also think it would be great 👍

Asmodeux2, 2024-10-26:
Админ, не ожидал тебя увидеть на GitHub.

bmorcelli, 2024-10-26:
It will come, in time

Teapot321, 2024-10-26:
> Это придет, со временем
Add it faster pls

IncursioHack, 2024-10-26:
Duplicated:
https://github.com/pr3y/Bruce/issues/216
https://github.com/pr3y/Bruce/pull/248

Petitoursbr1, 2024-10-29:
Just a question, what makes raw RF recording hard/slow to implement on Bruce ?

bmorcelli, 2024-10-29:
Time and priorities..

IncursioHack, 2024-10-29:
> Just a question, what makes raw RF recording hard/slow to implement on Bruce ?

Each implementation requires a number of tests, in addition, to make it work on various devices that we already have today (CC1101, RF433, FS1000...) and on various boards (Stick, Cardputer, ESP32-S3, Core, CYD, Lilygo... .). Another point is to do it in a way that does not impact other parts of the code.
Finally, there are priorities as @bmorcelli  mentioned.

Unfortunately, we dedicate ourselves to Bruce in the free moments we have and manage, we do not have the financial return to dedicate ourselves 100% to the project.

We invite you and everyone to develop for Bruce so that this process can be a little faster. Without this, unfortunately we are unable to carry out any implementation and/or changes in a timely manner.

Petitoursbr1, 2024-10-29:
I totally get that you can’t go all-in on this project since it’s not your main job, but honestly, what you’re doing already is awesome, and I’m extremely grateful

eadmaster, 2024-11-20:
For the **recording/receiving side**, getting a decoded dump of a signal if actually better than a raw dump, because the noise is filtered and so it is also more reliable to replay.

closed 2024-12-07 by bmorcelli