BruceDevices/firmware #373: Restarts at random

azrael-tzel, 2024-10-31:
**Describe the bug:**
Some times it restarts at random
And bleutooth spam seems to be working only some times

**Steps To Reproduce:**
- Device used: M5STICKC plus 1

**Expected behavior:**
BT Spam

**Actual behavior:**
Spammed a few times on my devices, but not much as Nemo. 

Restarts at random

**Additional context 
Will test it more and over and over, also on the go, but it seemed worth to send this in.

IncursioHack, 2024-10-31:
Regarding restarts, I also have a StickC 1.1, I use the Full firmware (without the Launcher), I haven't had any problems so far, but in your case we needed a log to understand what is happening, without it it is difficult to have this direction for analysis .

With reference to Spam, Bruce uses another Bluetooth library, this library allows (together with the code) other attacks such as BadBLE and the like. We understand that it loses a little performance compared to the old Nemo, at this point I don't think we will change it, the Spam gain is merely a prank, it has no effect of hacking or exploiting any serious flaw, considering that the version of IOS that crashed has already been fixed by Apple.

azrael-tzel, 2024-10-31:
Thanks alot for the reply, 

I will take it to the test a bit more, and in different simulated environments. 

I tought it was worth mentioning.

If i noticed something else i let you know, so far my compliments, it looks really promising.

azrael-tzel, 2024-10-31:
I will, maybe it was coincidence because right now all seems fine ( flashed it again )

IncursioHack, 2024-10-31:
Try to stay on the latest version available, if possible test the BETA version too, I suggest using the FULL version for M5StickC 1.1 (without the Launcher) and testing other features, for now I will close this issue, when you have any information about the restart, open it a new call with the Logs. Happy testing!

closed 2024-10-31 by IncursioHack