BruceDevices/firmware #375: Setting default AP name in EvilPortal

ventusz, 2024-10-31:
**Is your feature request related to a problem? Please describe.**
It's sort of a problem. When writing a custom evil portal html, how could I set a default wifi ssid for that portal?

**Describe the solution you'd like**
I have seen other projects that use a file named ap.config.txt in the same folder as the html file.

bmorcelli, 2024-10-31:
WiFi>Evil Portal>

Here you have two options, to choose standard (Google hadcoded) or custom html... If you choose custom, next step will be choosing a file

After that a keyboard will show up, written the default SSID (Free Wifi), that you can edit and choose whatever name you want.. if using Cardputer, try pressing backspace to erase, if using Stick or other, move the cursor to DEL and press M5 and write the desired Name there.

closed 2024-10-31 by bmorcelli

ventusz, 2024-10-31:
Thank you for your quick response.

I understand that you can do it like you mentioned. However I thought that maybe there was a way to pre-configure the AP SSID beforehand on the PC so that when I launch a captive portal it automatically uses that name.
Each portal could go in its own folder where it has the html file and the txt file which contains the SSID.

Could this be a new feature?