BruceDevices/firmware #376: CYD support

Teapot321, 2024-10-31:
Add ESP32-CYD-2432S024 support(CYD with Capactive touch)

IncursioHack, 2024-10-31:
https://github.com/witnessmenow/ESP32-Cheap-Yellow-Display/issues/41

IncursioHack, 2024-10-31:
Board name 	                Pixels 	                        Screen size 	                Touch type
ESP32-2432S024 	        2432 = 240x320 pixels 	024 = 2.4 inch LCD 	N/A
ESP32-2432S028R 	2432 = 240x320 pixels 	028 = 2.8 inch LCD 	R = Resistive
ESP32-3248S035C 	3248 = 320x480 pixels 	035 = 3.5 inch LCD 	C = Capacitive

We have 2 problems here, first is to actually check the name of the board and we recommend that when purchasing, please get a product that already has Bruce support, new implementations take time.

The second case is that we do not have the model you mentioned, so it is not possible to test the implementation.

Please provide us with more information regarding the board, its description, pinout and the like.

Teapot321, 2024-10-31:
<html>
<body>

ESP32-3248S035C | 3248 = 320x480 pixels | 035 = 3.5 inch LCD | C = Capacitive
-- | -- | -- | --

</body>
</html>

https://sl.aliexpress.ru/p?key=CBSzrd8

TheMiner203, 2024-10-31:
Hete is some info, link from Aliexpress page: http://pan.jczn1688.com/directlink/1/ESP32%20module/2.4inch_ESP32-2432S024.zip

closed 2024-11-01 by Teapot321