BruceDevices/firmware #379: CYD support (2.4+Capactive)

Teapot321, 2024-11-01:
I would like you to add support for the new CYD.
I have this CYD and I can't figure out what it is.
It has a 2.4screen as well as a Capactive touch, ESP32-2432S024 is written on the back, this CYD has a Type-C port. It has pins (GND, O35, O22, O21) x2 as well as Bat+ Bat- and one I2C. The packaging said ESP32-2432S024C

bmorcelli, 2024-11-01:
Do you have an example of project that uses this particular device?

Teapot321, 2024-11-01:
A link to the demo firmware (sort of) https://github.com/NoosaHydro/2.4inch_ESP32-2432S024/blob/main/1-Demo/Demo_Arduino/1_2_Factory_samples_Capacitive_touch/Factory_samples_Capacitive_touch/Factory_samples_Capacitive_touch.ino

Teapot321, 2024-11-01:
Here are the words of a person from a review on ALI:

Who will write the code, and not enjoy the built-in demo like an idiot. Touch CST820, screen ilI9341. NOT ilI9341_2!!!! The library for the CST820.h sensor is suitable for the TFT_eSPI screen. the User_config.ini file from the Chinese library does not need to be changed, it is in the capacitive version, in the resistive one. All the best, and success in development.

Teapot321, 2024-11-01:
I just installed this project and it works https://github.com/elik745i/ESP32-2432s024c

Teapot321, 2024-11-01:
Documentation from ali http://pan.jczn1688.com/directlink/1/ESP32%20module/2.4inch_ESP32-2432S024.zip
There are bin of demo

bmorcelli, 2024-11-01:
I will take a look on that device after I:
- Finish T-embed CC1101 port (make CC1101 and PN532 work)... It will provoke a new release
- change ./core files to make new ports easy, by creating an interface structure

So it will take a good time and effort, but making integration easy for other people to work on new devices

Teapot321, 2024-11-01:
[image]

bmorcelli, 2024-11-16:
Added here https://github.com/pr3y/Bruce/pull/427

closed 2024-11-16 by bmorcelli