BruceDevices/firmware #381: RF Spectrum bug

Teapot321, 2024-11-01:
I tried the spectrum on two keys and it won't work. I went into Scan/Copy and when All ranges is selected, the frequency is determined when the key is pressed.
I tried spectrum at CatHack and it works there.

Teapot321, 2024-11-01:
I think I've figured out what the problem is.
https://github.com/pr3y/Bruce/issues/249

bmorcelli, 2024-11-01:
Oooohhh you are using SadCard connected with the CC1101... It won't work, sorry 

For it to work, you need to use other pin setting, with a modded firmware, a transistor and a few resistors (and a dip switch if you want to have NRF24 with it in the same circuit)

I ordered a board made by Th3_Kr4k3n, to test.. if it works well, we will share the schematics and make it official..

But for now.. SDCard with NRF24 and/Or CC1101 is not Supported.

closed 2024-11-01 by bmorcelli