BruceDevices/firmware #382: 7 inch cyd support

mattbad04, 2024-11-01:
Would love to have bruce working on my 7 inch cyd 
esp32-8048s070 ive found it has a different screen controller then the others any support would be greatly appreciated !!!

bmorcelli, 2024-11-02:
If you can share an example of a project that works on this particular device, it can speed up the process, as I (we) don't have this board im hands to test.

I will take a look on that device after I:
- Finish T-embed CC1101 port (make CC1101 and PN532 work)... It will provoke a new release
- change ./core files to make new ports easy, by creating an interface structure

So it will take a good time and effort, but making integration easy for other people to work on new devices

mattbad04, 2024-11-02:
I'm in town at the moment but I will get as much information on the device and put it here also if you give me a shipping address I can ship you one aswell I accidentally bought 2 of them

mattbad04, 2024-11-02:
[image]

mattbad04, 2024-11-02:
[image]

mattbad04, 2024-11-02:
https://nightly.openhasp.com/

mattbad04, 2024-11-02:
https://www.openhasp.com/0.7.0/hardware/elecrow/crowpanel-hmi-rgb/

mattbad04, 2024-11-02:
[image]

mattbad04, 2024-11-02:
I have 3 panels two is a Sutton esp32 s3 and the other is the crowpanel by elecrow all are 7 inch I do believe each brand uses different micro controllers I have a panel I can spare that yall can use for the project

mattbad04, 2024-11-02:
[image]

mattbad04, 2024-11-16:
#define TOUCH_MODULES_GT911
 #define TOUCH_MODULE_ADDR GT911_SLAVE_ADDRESS1
 #define TOUCH_SCL 20
 #define TOUCH_SDA 19
 #define TOUCH_RES 38
 #define TOUCH_INT -1

mattbad04, 2024-11-16:
// Please fill below values from Arduino_GFX Example - TouchCalibration
bool touch_swap_xy = false;
int16_t touch_map_x1 = 0;
int16_t touch_map_x2 = 800;
int16_t touch_map_y1 = 0;
int16_t touch_map_y2 = 480;
[11:32 PM]
/Show some widget. It might be required to increase LV_MEM_SIZE/
#define LV_USE_DEMO_WIDGETS 1

mattbad04, 2024-11-16:
[image]

mattbad04, 2024-11-16:
[image]

mattbad04, 2024-11-16:
[image]

mattbad04, 2025-02-02:
Any help for progression on this board will be much appreciated thank you i will also donate!!!

XeroX42004, 2025-05-07:
Continuing my efforts tomorrow!