BruceDevices/firmware #383: Bruce suggestions form a long time user

Danyokkk, 2024-11-02:
Hello,

I have used Bruce already for a long time and would like you to suggest some improvements.

1. Improve SubGhz - make the signal better and try to make it compatible with Flipper to transmit .sub files from Flipper Zero.
2. IR Send - send recorded IR codes or codes from Flipper Zero.
3. SD card - make Bruce more compatible with SD card to save signals and then send them.
4. NFC/RFID - read: read NFC/RFID tags and save data to SD card, NFC/RFID emulate: **emulate** NFC/RFID tags using saved data.

For now that all, your software is amazing. For now, your software is n1 but still needs some improvement.

Thank you!

rennancockles, 2024-11-02:
I don't know if you've actually been using Bruce for a long time, since a lot of the things you're asking for are already there. So I think you should flash the latest version from our site and test it out a bit more.

Danyokkk, 2024-11-02:
Thanks for the reply, I have used it for a pretty long time but your software misses some of the functions like:

- RFID emulating
- SD Card (not working with CC1101 and NRF24) how can we record signals and send them back with these modules?
- .sub files are not compatible with Bruce which it makes not very comfortable if you want to combine Bruce software with Flipper Zero or just send .sub files to your m5stick.
- CC1101 and SD Card problem
- SubGhz doesn't have a RAW protocol

Jethro-San, 2024-11-02:
Most of those features already exist, and if you want to have these added you can just find someone else's esp32 firmware that does this and give us the link. that makes it much easier to incorporate. (Even better is code it yourself!).

Danyokkk, 2024-11-02:
I don't have that much coding experience and I have already searched for more firmware but yours is the best (especially for m5stick it makes a real beast out of it) your firmware just needs some further improvements. My message to you is just my **suggestions** for future updates that will make your firmware even more **powerful**.

rennancockles, 2024-11-02:
Suggestions are always welcome, the thing here is that you're requesting features that already exist or are impossible to make.

- RFID emulating is possible using a chameleon device. There's no board with emulation capabilities for us to develop for it (if you know any just let me know)
- SD card does work with cc1101 and nrf24. It's all fine with cardputer and with sticks you just have to remove and insert the sd card again because they share the same SPI bus.
- Bruce is compatible with flipper's .sub files. Maybe not all protocols, but the files share the same format.
- problem with cc1101 and sd card was mentioned on the item 2
- About the RAW protocol I'm not sure if we support it or not yet. You might be right about this one.

Danyokkk, 2024-11-03:
Thanks for explaining all this in more detail.

- RFID Emulating: I didn’t know it needed a chameleon device, and unfortunately, I can’t test with one. I'll let you know if I find a board that can be used for emulation.
- SD Card with CC1101 and NRF24: I’ll try reinserting the SD card as you suggested, though I don’t have much experience troubleshooting hardware issues. I hope it will work.
- .sub Files Compatibility: Good to know Bruce is compatible with .sub files. I might not have tested enough protocols yet, so I'll keep that in mind.
- RAW Protocol: It would help if you could confirm whether there is any support or plans for the RAW protocol.

Thank you

IncursioHack, 2024-11-03:
> Maybe also try to add a **PN5180** module support with Bruce because as I know it can emulate signals.

I didn't find concrete documentation or code with reference to emulation, some pages have small descriptions of what it would be capable of emulating, but I understand that it is the same case as the PN532, where it is not possible to emulate in a practical way. Another point is the cards and frequency that this module accepts. Chameleon supports several models and also supports the 2 frequencies 13.56mhz (HF) and 125khz (LF), this has been the best option so far (in addition to being professional equipment). I left some models below:

https://encurtador.com.br/3phCl

> RAW Protocol: It would help if you could confirm whether there is any support or plans for the RAW protocol.

No expected implementation date.

Danyokkk, 2024-11-03:
Thanks for your thoughts!

I looked into the PN5180, and it seems like it can emulate but there is not that much info about it and I agree that PN532 is better and easier for now but I found some PDFs that have some information about PN5180, and code examples for it which you can look at.

- https://www.nxp.com/docs/en/user-guide/UM10954.pdf
- https://www.farnell.com/datasheets/2148709.pdf

Also, about the Chameleon, yes it supports several models and also supports the 2 frequencies 13.56mhz (HF) and 125khz (LF) but it's pretty expensive so it would be very nice if we could find a way to make emulating on Bruce using cheaper modules.

IncursioHack, 2024-11-03:
I believe it is the same scenario as the PN532, there is some lost documentation about emulation, but it is not possible to emulate all bytes, one of the bytes (the 1st) is written in the code and it is not possible to change it, making emulation unfeasible. I looked for @rennan and did several tests and research for months and we didn't find any module that is capable of emulating in a cheap way and that could be integrated into Bruce in a practical way. In the link I sent you there is a version of Chameleon for around $16.

Danyokkk, 2024-11-03:
I see, so the $16 version of Chameleon is working with m5stick and can it emulate? Can you please send me a link to this Chameleon version because I can't find it in the link you send me.

IncursioHack, 2024-11-03:
> I see, so the $16 version of Chameleon is working with m5stick and can it emulate? Can you please send me a link to this Chameleon version because I can't find it in the link you send me.

Chameleon Ultra (RFID/NFC/Emulate): https://s.click.aliexpress.com/e/_mN6vZiG

IncursioHack, 2024-11-03:
> I see, so the $16 version of Chameleon is working with m5stick and can it emulate? Can you please send me a link to this Chameleon version because I can't find it in the link you send me.

Its connection is via Bluetooth, you can connect it via Stick, Cardputer, CYD and other cards that already support Bruce :)
https://www.youtube.com/watch?v=ZInyYs_0YAw

Danyokkk, 2024-11-03:
Thank you, I will check it out

eadmaster, 2024-11-04:
> RAW Protocol: It would help if you could confirm whether there is any support or plans for the RAW protocol.

Should be supported with both IR and RF, but it needs more testing: https://github.com/pr3y/Bruce/issues/216

Petitoursbr1, 2024-11-04:
I just want to know how can you access thoses functionalities even if they are in early development as I want to try them out to maybe give my support or feedback is there a link or script that I can download to make it work ?

IncursioHack, 2024-11-04:
> I just want to know how can you access thoses functionalities even if they are in early development as I want to try them out to maybe give my support or feedback is there a link or script that I can download to make it work ?

If it is an issue on github, it has probably not been developed yet or is only on the developer's machine while it is still being created. Now, if a Pull Request (PR) has already been made to the code and it has been committed, it becomes available to use in the BETA version ( test) on the website https://bruce.computer  > Install > Beta Release

eadmaster, 2024-11-04:
> I just want to know how can you access thoses functionalities even if they are in early development as I want to try them out to maybe give my support or feedback is there a link or script that I can download to make it work ?

you can test raw **receiving** with [serial cmds](https://github.com/pr3y/Bruce/wiki/Serial) `ir rx raw` and `ir tx raw`.
you can test raw **sending** simply by loading `ir/sub` files  using that protocol.

Ademolght, 2024-11-04:
@rennancockles @IncursioHack i think @Danyokkk is blocked in v0.8 of bruce 😂

> Hello,
>
> I have used Bruce already for a long time and would like you to suggest some improvements.
>
> 1. Improve SubGhz - make the signal better and try to make it compatible with Flipper to transmit [...]

bmorcelli, 2024-11-14:
Esp32 have a limited flash and RAM memory to use... 

It already have the iconic TV-b-gone function from this [repo](https://github.com/agrimpelhuber/esp8266-tvbgone)

And it has the capacity to run custom commands from any .ir file (some of them with these universal commands) that you can get in [this repo](https://github.com/Lucaslhm/Flipper-IRDB)

Danyokkk, 2024-11-14:
> Esp32 have a limited flash and RAM memory to use...
>
> It already have the iconic TV-b-gone function from this [repo](https://github.com/agrimpelhuber/esp8266-tvbgone)
>
> And it has the capacity to run [...]

Thank you, checked it out, do you know repo for SubGHz files?

closed 2024-12-07 by bmorcelli