BruceDevices/firmware #384: device support

cinouw, 2024-11-02:
lilygo t display s3 support would a good addition to supported devices list

bmorcelli, 2024-11-03:
There are 3 models of T-Display

* Non touchscreen, with only 2 btns (probably won't be ported due to lack of btns, at least 3 are desired)
* T-Display S3 Touch, can be done
* T-Display S3 Pro, can be done

cinouw, 2024-11-04:
> There are 3 models of T-Display
> 
> * Non touchscreen, with only 2 btns (probably won't be ported due to lack of btns, at least 3 are desired)
> * T-Display S3 Touch, can be done
> * T-Display S3 Pro, can be done
 
Hi I have the version with only 2 buttons  I think it's doable with connecting a button to the gpio pins 

If you think of doing it I can be the tester if you want 

Thanks in advance

bmorcelli, 2024-11-04:
I'm working on interfacing devices into Bruce, and maybe in the future we come up with a keyboard solution and navigation for 2btn devices..

I agree that adding btns and stuff to these boards is easy, but the harder part is give support to those who want to use the board, so I won't port for now due to this reason, but don't loose your faith ;)

cinouw, 2024-11-05:
> I'm working on interfacing devices into Bruce, and maybe in the future we come up with a keyboard solution and navigation for 2btn devices..
>
> I agree that adding btns and stuff to these boards is [...]

;) I'll be always here

d4rks1d33, 2025-01-06:
Hi I have a t-display to (not the S3 but essentially is the same but without touchscreen) I would love to help port Bruce into this devices (t-display and t-display S3) but I stuck in the way you build the firmware (I more Arduino IDE guy lol) but if somehow help I could test the firmware if u want to

bmorcelli, 2025-01-06:
@d4rks1d33 https://github.com/pr3y/Bruce/wiki/Building-from-source#vs-code