BruceDevices/firmware #394: Gps not working

Trhug25, 2024-11-06:
I have a 6m neo gps module. I have connected the pins as follows 
Vcc - vin
Rx - rx
Tx - tx
Gnd - gnd 
I connected such pins to cheap gold screen cyd 2 usb . But it seems gps does not work, gps waiting time up to 700s but nothing happens.

rennancockles, 2024-11-06:
Try switching Rx and Tx pins

Trhug25, 2024-11-06:
> Hãy thử thay đổi chân Rx và Tx

I tried. It still doesn't work.

bmorcelli, 2024-11-06:
Gps doesn't work on CYD yet

Trhug25, 2024-11-06:
> Gps vẫn chưa hoạt động trên CYD

Oh. Come to me to run it up to 12000s

Trhug25, 2024-11-06:
> Gps vẫn chưa hoạt động trên CYD

https://github.com/justcallmekoko/ESP32Marauder/wiki/gps-modification
I wired this project.

Trhug25, 2024-11-06:
[image]
[image]
[image]
N
[image]
It works when I use io2 
The connection diagram is as follows 
Gnd - gnd
Io22 - tx
Io27 - rx 
3v3 - vcc

closed 2024-11-07 by Trhug25