BruceDevices/firmware #402: RF scan option error 1.6.2

akali997, 2024-11-08:
hello, i am new to m5card as well as bruce firmware, i want to use cc1101 with bruce, i soldered the wire and edited it a little bit, added a socket pin suitable for cc1101 on the side, i tested other functions like spectrum or simply i put the frequency back to all, but i tried with my door key it only shows the frequency, so i am sure that my soldering part has no problem, i think it is due to the firmware, please fix it as soon as possible in a newer version, or someone tell me which version works stably, thanks
[image]
[image]

akali997, 2024-11-08:
[image]
Đây là màn hình hiển thị lúc tôi ấn điều khiển rf , nó chỉ ở đó , hoặc đơn giản là nhận ra tần số là 330

akali997, 2024-11-08:
[image]
This is the frequency display when selecting scan/copy, the only difference is that it displays the correct frequency of my control, and does not display anything else.

IncursioHack, 2024-11-08:
Your key modulation may not be compatible. In addition to frequency, there are several factors such as modulation, interference, encryption, rolling code and the like. Bruce predicts a specific scenario and if I'm not mistaken with ASK/OOK modulation, if your key is different from that, it is possible that he will not identify this data, only that "something" is transferring "something" at a specific frequency.

Radio Frequency has a very large universe, some friends have an amateur radio certificate, they have lived in this universe for 20 years and don't know the half of it, so it ends up being more difficult for someone to gain knowledge in this part and especially to predict various scenarios.

This is why Spectrum captures "something" most of the time, but it has no idea what it is, for comparison purposes it's as if we went to Yugoslavia and people were speaking Slovenian, you know something is being said (spectrum ) but has no idea what it is, the content and how to separate important parts (scan).

What I explained above is just the surface of everything it can be. Without having the key exactly the same as yours to analyze/test, it is almost impossible to say that it will be decoded via software. What could happen in the future is to capture this signal in RawCode and store it in the SD, however, it will not be presented on the screen decoded (separated and identified).

Some reference issues:
https://github.com/pr3y/Bruce/issues/286

akali997, 2024-11-11:
thank for rep, i will try with other remote or rf module

bmorcelli, 2024-11-21:
Can you test now with version 1.7.2? 

Bruce can read raw data now, ao maybe you can get something

closed 2024-12-07 by bmorcelli