BruceDevices/firmware #403: changes and additions

loni1488, 2024-11-08:
1. The color of the theme does not change in the spectrum, but remains purple

2. during Wi-Fi attacks, show a list of the names of the networks that are being jammed

3. I looked at the firmware code and saw that some spam dots in the ble just go as text and not a command

4. add the possibility of a simple mp4 video (I do not know if this is possible or not, if not, then sorry for the stupid request)

QWEIVAN, 2024-11-08:
Add new copy rf protocol (raw)pls

IncursioHack, 2024-11-08:
> 1. The color of the theme does not change in the spectrum, but remains purple
>     2. during Wi-Fi attacks, show a list of the names of the networks that are being jammed
>     3. I looked at the [...]

1. I think it's possible, when I have time I'll take a look.
2. If you are talking about the NRF24 (Jammer), this is not possible.
3. Please give me examples or send me a PR with the fix if it is indeed a problem.
4. I don't think it's possible, ESP32 is absurdly limited.

> Add new copy rf protocol (raw)pls

We are thinking about it, with no prediction as to when it will be implemented.

QWEIVAN, 2024-11-08:
thank you, bro

loni1488, 2024-11-09:
2. No, mean wifi atks

QWEIVAN, 2024-11-20:
please fix the search for fm module on m5stack stick / cardputer because this is a very useful function, or make detailed connection instructions

closed 2024-12-07 by bmorcelli