BruceDevices/firmware #408: Question : Device support sugestion

Z-SHOCK-E, 2024-11-10:
I came across this dev board that I think will be great for this firmware. It has the esp32-s3 chipset (I think it has USB_HID), a display and more gpio (for the cc1101 and the nrf24l01). 

https://www.aliexpress.com/item/1005006454900498.html?spm=a2g0o.productlist.main.3.337f6d78rGBHB5&algo_pvid=82830486-2dcd-4f04-9997-99e434c96d64&algo_exp_id=82830486-2dcd-4f04-9997-99e434c96d64-1&pdp_npi=4%40dis%21LKR%212283.15%212283.15%21%21%217.80%217.80%21%40212e532617312298597756893eff56%2112000037259796603%21sea%21LK%212397466382%21X&curPageLogUid=you6MM1hzMIf&utparam-url=scene%3Asearch%7Cquery_from%3A

bmorcelli, 2024-11-11:
The big challenge for the actual code is the navigation with only 2 btns..

As I'm developing an "interface" logic to develop the ports for the devices, maybe (MAYBE) it'll be possible to think on a way to make the core functions to work in this situation (2 btns only)

I know that the user can add more btns, but I prefere working on less hw changes as possible, so the code is plug and play for everyone.

Z-SHOCK-E, 2024-11-14:
Ok. Thanks

closed 2024-11-14 by Z-SHOCK-E