BruceDevices/firmware #410: Is a UI refresh ever planned?

bisaw37, 2024-11-12:
This isn't really on any priority list, but it seems wasteful to have such a nice full color and decent resolution display (on basically all of the m5stack products at least) and not really use it. Having a one color sharp and blocky UI seems a bit bad to me at least, so I want to know if eventually, a UI redo is planned. I'd say to give a color to each function, like red for IR, purple for RF, green for RFID, blue for NRF, and so on. Also making the menus look more comprehensive, for example making menus like the one displayed when using IR record, look nicer and use more of that good multicolor display. Rounded corners, different shades, complementary colors, the sort.

rennancockles, 2024-11-12:
No, it's not on our plans. Bruce is not made for m5stack's devices only. It was made for any esp32 based devices, so we need to keep compatibility with many different screens. Maybe we can improve a few things in the future, but it's not a priority. However you can contribute and develop a new theme for Bruce if you like. Feel free to create a PR and contribute with the project.

bmorcelli, 2024-11-16:
No

closed 2024-11-16 by bmorcelli