BruceDevices/firmware #416: M5StickC Pluse 2. Change GPIO pin from G33 (Grove SCL) to G22 (SCL)

dgorbatko, 2024-11-13:
Hi, I tried to modify the board of M5StickC Pluse 2 and accidentially damaged pin 33 (Grove SCL/Groove W) It just fell off from the board. 

1. I found an empty pin with the word "CSL" on the board.
With help of Chat gpt i found that this pin is 22.
Then I loaded a simple pin test code via arduino ide like pin 22 high 1000ms, and then pin 22 low 1000ms.
Tested it with multimetr and saw that the voltage is jumping from 0 to 3.2 as it should be.
Then I opened your source code and changed 33 to 22 whereever I could find in Platformio.ini file.
But after compiling and burning the firmware i have constand 3.2v on this pin whatever I load there (IR, NRF, CC1101 etc).

What else has been tried:
2. I desolder buzzer to take it's gpio 2, but there is only 0-0.8V maximum via the pin test code.
3. I desolder the inbuilt IR to take it's gpio 19 but there is also only 0-0.8V during HIGH load during pin test code.
4. I also tried to find the trace from broken groove pin 33 but no luck. I scratched the paint around the pin but there is only ground everywhere.

So i see the only one way. Is to try to use CSL PIN 22 as a replacement from grove 33 but it needs to be tweaked in the frimware, bcs it looks like something is sitting on this pin somewhere else and make it always 3.2v. And I understand that future your updates I need to manually tweak before update my m5stick.
If you could point me where to change something in the firmware or give some ideas what to do, because without this pin m5stick is pointless (there is no way to connect nrf24 to it) so I want to try at least something before throwing it to trash.

Because I believe its possible because on the pin test simple code this pin works as a simple Gpio (it looks like).

Thanks!

bmorcelli, 2024-11-13:
This pin is used by RTC chip

https://docs.m5stack.com/en/core/M5StickC%20PLUS2

You can try disabling the RTC chip to see if it works, commenting/deleting this flag here:
https://github.com/pr3y/Bruce/blob/de50cff22b5c0c26421fa492109c895f2acb59a2/platformio.ini#L110

But I would suggest you to buy a replacement for the Grove HY2.0-4P male input 
https://a.aliexpress.com/_mrmoW6X

Hope you can fix it.

dgorbatko, 2024-11-13:
> disabling the RTC chip

Yes it works!
Thank you very much!

Unfortunately I can't resolder the HY20 port because I don't have g33 pin anymore at all, it just fell off the board during the unsoldering. I wanted to move HY ports to the main 8pin socket. Replace battery to bigger one and made buttons more clickable. And I already redesigned and 3dprinted a new body for it. But failed on g33 pin during unsoldering. Now I can finish my project!

Next step I want to try to edit your NRF24 jammer tool to let it works with dual NRF24 on different channels at the same time for better result. Did you try it before?
Like here: https://github.com/smoochiee/Bluetooth-jammer-esp32

bmorcelli, 2024-11-13:
You doesn't have enough gpio for that..

You would need one more pin for it to work, to use as CS for the second nrf24 (otherwise you won't be able to set up the channels separately).. CE pin can be shared between them