BruceDevices/firmware #417: Keyboard bug on CYD-2432S028R

TheMiner203, 2024-11-14:
**Describe the bug:**
When you press a button on the touch keyboard, not one character is entered, but several at once

**Steps To Reproduce:**
Write something on the touch keyboard (for example, a Wi-Fi password)
- Device used: CYD-2432S028R

**Expected behavior:**
First, one character is entered, after about a second (the time to release the button), the input begins to repeat

**Actual behavior:**
With a short press, several characters are entered, without waiting for the release

bmorcelli, 2024-11-14:
Hey friend, can you test the BetaRelease?

I think I have already fixed it [here](https://github.com/pr3y/Bruce/commit/e1d32c1edcf70bd42f9d9a563c19206885fc53c6)

TheMiner203, 2024-11-14:
> Hey friend, can you test the BetaRelease?
> 
> I think I have already fixed it [here](https://github.com/pr3y/Bruce/commit/e1d32c1edcf70bd42f9d9a563c19206885fc53c6)

On BetaRelease, the firmware does not work at all, it just endlessly tries to connect to WiFi without letting you click anything. I can record a video if necessary

flaryx32, 2024-11-14:
> On BetaRelease, the firmware does not work at all, it just endlessly tries to connect to WiFi without letting you click anything. I can record a video if necessary

same issue on beta :/

bmorcelli, 2024-11-14:
Try the last beta version..

It has a "debounce" of 200ms.. holding the key will add more letters, and it is intentional..

TheMiner203, 2024-11-15:
Everything works as it should in the latest beta:)

closed 2024-11-15 by TheMiner203