BruceDevices/firmware #432: Which TFT display can be used with the esp32-s3 devkit. How to connect the display?

Z-SHOCK-E, 2024-11-16:
Which tft display will work with bruce without a lot of additional work. 1.9" and 2.8" TFT displays are already implemented in the t_embed and the cyd.

bmorcelli, 2024-11-19:
all displays compatible with TFT_eSPI library can be ported to Bruce.

```
Displays using the following controllers are supported:

GC9A01
ILI9163
ILI9225
ILI9341
ILI9342
ILI9481 (DMA not supported with SPI)
ILI9486 (DMA not supported with SPI)
ILI9488 (DMA not supported with SPI)
HX8357B (16-bit parallel tested with RP2040)
HX8357C (16-bit parallel tested with RP2040)
HX8357D
R61581
RM68120 (support files added but untested)
RM68140
S6D02A1
SSD1351
SSD1963 (this controller only has a parallel interface option)
ST7735
ST7789
ST7796
```

closed 2024-11-19 by bmorcelli

samallx, 2024-11-21:
can you please make a guide how to use esp 32 wroom + ili9341, i beg you