BruceDevices/firmware #439: Bad BLE Ducky Script on M5Stick C Plus 1.1 doesn't work on Windows 11.

logm1lo, 2024-11-18:
**Describe the bug:**
When I tried to use the Bad BLE Feature on the M5Stick C Plus 1.1 using the dev release (I built it). Everything in Media CMDS works but when I used the ducky script, it doesn't run on my laptop.

**Steps To Reproduce:**
Build the latest build, use the Bad BLE,connect the m5stick to the laptop via bluetooth, use this ducky script: [The_Matrix_Wake_Up.txt](https://github.com/user-attachments/files/17801041/The_Matrix_Wake_Up.txt)

- Device used: M5Stick C Plus 1.1, Windows 11

**Expected behavior:**
The laptop executes the script

**Actual behavior:**
The laptop doesn't executes the script

bmorcelli, 2024-11-18:
Make sure your stickCPlus2 is not connected to 2 devices at the same time  (smartphone and computer)

logm1lo, 2024-11-19:
> Make sure your stickCPlus2 is not connected to 2 devices at the same time (smartphone and computer)

I am sure I didn't connect to the smartphone

logm1lo, 2024-11-20:
Even though I isolated my smartphone by putting it in the faraday bag, the Bad BLE still doesn't run Ducky script on my windows 11 computer.

logm1lo, 2024-12-05:
> For me on the same devices mentioned it works perfectly.

that's weird, did you compiled it yourself or did you used the prebuilt one?

logm1lo, 2024-12-07:
> both

well, for me, it's still doesn't do anything

closed 2024-12-07 by bmorcelli