BruceDevices/firmware #447: Display color inverted on ESP32-2432S024C

Asmodeux2, 2024-11-19:
**Describe the bug:**

**Steps To Reproduce:**

- Device used: ESP32-2432S024C 

[image]

Asmodeux2, 2024-11-19:
Use firmware CYD-2432W328C.
Touch screen work perfect!

Asmodeux2, 2024-11-19:
[image]

Asmodeux2, 2024-11-19:
ports/CYD-2432S028/platformio.ini - 

-DTFT_INVERTION OFF

bmorcelli, 2024-11-19:
It is better to have 2 versions, one with inverted colors, because who developed ot made sure his version did work with appropriate color

ciccirix, 2024-11-20:
dove trovo i file per  ESP32-2432S024 ?

ciccirix, 2024-11-20:
si possono aggiungere su https://bruce.computer/flasher.html?

Asmodeux2, 2024-11-20:
> dove trovo i file per  ESP32-2432S024 ?
> 

Use firmware CYD-2432W328C

ciccirix, 2024-11-20:
> Use firmware CYD-2432W328C on Release

Dont work… white screen 😂

Asmodeux2, 2024-11-20:
> Dont work… white screen!!
> 
> 
> > dove trovo i file per ESP32-2432S024 ?
> >
> > Use firmware CYD-2432W328C on Release
> >
> > <https://github.com/pr3y/Bruce/issues/447#issuecomment-2487563382>, or
> > unsubscribe

Look closely it works, just the colors are inverted and black became white and vice versa.  Now I compile a corrected version for ESP32-2432S024C

ciccirix, 2024-11-20:
HO PROVATO LE DUE VERSIONI SU https://bruce.computer/flasher.html MA NIENTE SCHERMO MORTO E RED ROSSO

Asmodeux2, 2024-11-20:
> HO PROVATO LE DUE VERSIONI SU https://bruce.computer/flasher.html MA NIENTE SCHERMO MORTO E RED ROSSO

Shot picture under 5 minutes on

ciccirix, 2024-11-20:
[image]
ONLY 1 USB C VERSION

Asmodeux2, 2024-11-20:
> 
> https://github.com/user-attachments/assets/bf6468a3-051f-431a-b0ea-c06fac735ac6
> 
> 
 Man, you misunderstand me. You have a version of ESP32-2432S024R. Resistive touch. In your case you just need to change the screen size of ESP32-2432S028 firmware. And don't use WebFlasher, it won't work for you anyway.

Asmodeux2, 2024-11-20:
> [image]
> ONLY 1 USB C VERSION 

https://github.com/Asmodeux2/ESP32Marauder_ESP32-2432S024R_CYD - install this ESP32 Marauder for you CYD

Asmodeux2, 2024-11-20:
> https://github.com/pr3y/Bruce/blob/main/ports%2FCYD-2432S028%2Fplatformio.ini#L181
> 
> Change pls OFF

[image]

After changing this value to OFF. All colors are perfect and everything works perfectly on ESP32-2432S024C

pdx-rico, 2024-11-20:
> > https://github.com/pr3y/Bruce/blob/main/ports%2FCYD-2432S028%2Fplatformio.ini#L181
> > Change pls OFF
> 
> ![20241120_104006.jpg](https://private-user-images.githubusercontent.com/136321449/387994729-4275db2e-a8f8-45b7-922a-05156e6cbfed.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzIxMTYxMzQsIm5iZiI6MTczMjExNTgzNCwicGF0aCI6Ii8xMzYzMjE0NDkvMzg3OTk0NzI5LTQyNzVkYjJlLWE4ZjgtNDViNy05MjJhLTA1MTU2ZTZjYmZlZC5qcGc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMTIwJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTEyMFQxNTE3MTRaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1mYTRhNGM2ZGE2M2Q2NTFjMTQ4ZjY2NGZhNjA4YjlmYjYzNDI3YjY5OTUyNDgyZmY5NDFmNjFlZDMxMzI1MDA3JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.RlIMTeQ5uTC-jGnkLGrm0BeRrWCzE0VNNuYQZIXr0-4)
> 
> After changing this value to OFF. All colors are perfect and everything works perfectly on ESP32-2432S024C

Hey, m8! I'll have same CYD(ESP32-2432S024C), please help how can i write this firmware on it? Any help appreciated

Asmodeux2, 2024-11-20:
> Hey, m8! I'll have same CYD(ESP32-2432S024C), please help how can i write this firmware on it? Any help appreciated

Write to me in Telegram @Ud7egwistv33u, I will send you ready firmware for ESP32-2432S024C. I honestly don't understand much in this area of microcontrollers myself.

closed 2024-11-20 by Asmodeux2

mrbungle911-beep, 2026-02-16:
does anyone have the modified value version available