BruceDevices/firmware #449: Bruce versione 1.7 problema con il modulo CC1101

Luigi-77, 2024-11-20:
Nel menu RF in modalità scan/copy il dispositivo sembra impazzito come se ricevesse un codice in maniera continuativa sia con il dispositivo M5Cardputer e con CYD con 2 porte USB e non viene visualizzato nessun codice binario.

bmorcelli, 2024-11-21:
RF is now receiving RAW data, so if it can't decode, it will detect and be able to save the raw data to replay when needed

Can you please try the version 1.7.2 that was just released?

Luigi-77, 2024-11-21:
Ok grazie

BeqaMiqa, 2024-11-23:
> Salve, dove trovo il pinout per cc1101?? uso cyd 2.8

Is this what you are looking for 
https://github.com/pr3y/Bruce/wiki/CC1101

It's at the bottom.

And also does anyone know if CC1101 work with the cyd micro USB versions?

closed 2024-11-26 by bmorcelli