BruceDevices/firmware #450: A new useful feature

Teapot321, 2024-11-20:
Make it possible to change the presence of partitions in the settings/configuration file.
Let's say I want to remove RFID, FM, Scripts and Connect. This is what bruce.conf looks like with the following parameters:

WiFi--1
BLE--1
RF--1
RFID--0
IR--True
FM--0
NRF24--1
Scripts--0
Clock--1
Connect--0

bmorcelli, 2024-11-20:
It is open source, you can comment these lines, or change the order if you will, and build it for your own purposes..

https://github.com/pr3y/Bruce/blob/main/src%2Fcore%2Fmain_menu.cpp#L7-L26

Adding this kind of feature adds a whole unnecessary complexity to the project which doesn't add in feature (skipping things, avoiding clicks?)

eadmaster, 2024-11-20:
Duplicate of https://github.com/pr3y/Bruce/issues/326