BruceDevices/firmware #455: Add Inversion Toggle Feature in Config

Asmodeux2, 2024-11-20:
---

### Description:

**Background:**
Currently, to enable or disable inversion, we need to create separate firmware images with the flags `-DTFT_INVERSION_ON` or `-DTFT_INVERSION_OFF`. This approach leads to multiple firmware images for a single feature, which is inefficient and cumbersome.

**Proposal:**
To streamline the process and provide more flexibility to the users, we propose adding an inversion toggle feature directly in the Config section of the firmware. This way, users can easily enable or disable inversion without the need for multiple firmware images.

**Benefits:**
1. **Simplified User Experience:** Users can toggle inversion on or off directly from the Config menu, making the firmware more user-friendly.
2. **Reduced Firmware Variants:** Eliminates the need to maintain and distribute multiple firmware images for a single feature.
3. **Efficient Development:** Simplifies the development and testing process by reducing the number of firmware variants.

**Implementation Steps:**
1. Add a new configuration option in the firmware settings menu to toggle inversion on or off.
2. Update the firmware code to read this configuration option and apply the corresponding inversion setting.
3. Test the new feature thoroughly to ensure it works as expected and does not introduce any new issues.

**Expected Outcome:**
Users will have a more flexible and user-friendly way to control the inversion feature, and the development team will have a simpler and more efficient process for managing firmware variants.

**Additional Notes:**
- Ensure that the new configuration option is well-documented in the user manual and any relevant documentation.
- Consider adding a default setting for the inversion toggle to maintain backward compatibility.

---

### Labels:
- enhancement
- feature-request
- config

### Assignees:
- @Bruce

### Milestone:
- Next Release

---

Please review and provide feedback or any additional suggestions.

Best regards,
Asmodeux2

bmorcelli, 2024-11-20:
> Assignees

Hhahahaha

Resquest like these make me thirsty for many cups of  [coffee](https://buymeacoffee.com/bmorcelliz)

Asmodeux2, 2024-11-20:
> > Assignees
> 
> 
> Hhahahaha
> 
> Resquest like these make me thirsty for many cups of  [coffee](https://buymeacoffee.com/bmorcelliz) 

I'd love to give you many mugs of coffee, but the problem is that I'm from Russia.[image]

bmorcelli, 2024-11-21:
Please test flashing this version

[image]

Asmodeux2, 2024-11-21:
All work

Asmodeux2, 2024-11-21:
Rename pls to ESP32-2432S024C @bmorcelli

bmorcelli, 2024-11-21:
But that is for 2.8" display (at least in the code)

Asmodeux2, 2024-11-21:
> But that is for 2.8" display (at least in the code)
> Edit: the name you want is for 2.4" display, which Idk if works or not
> 
> 

But it works fine on the ESP32-2432S024C. And many people in Russia use ESP32-2432S028, ESP32-2432S024C

Asmodeux2, 2024-11-21:
> But that is for 2.8" display (at least in the code)
> Edit: the name you want is for 2.4" display, which Idk if works or not
> 
> 

It also integrates a touch screen support library with the CST820 module, which also works perfectly. My friend @Teapot321 and many others who have installed the firmware can confirm this.

Asmodeux2, 2024-11-21:
@bmorcelli Sample rename CYD-2432W328C_2 -> CYD-2432S024C

bmorcelli, 2024-11-21:
But that is for 2.8" display (at least in the code)

Calm brother, it is building the flasher, take it easy

Asmodeux2, 2024-11-21:
Okay, man, you know better. And you're pretty cool, too.👍🏻

closed 2024-11-26 by bmorcelli