BruceDevices/firmware #458: M5StickCPlus2 Flash Tool Error

ATOMNFT, 2024-11-21:
**Upon connecting and then trying to install I'm presented with a error that states "Failed to download manifest"**

**Steps To Reproduce:**

- Device used: M5StickCPlus2 
- Try to use web flasher to flash latest bruce to M5StickCPlus2

**Expected behavior:**
Flash tool to start the flashing process and me to wait. 

**Actual behavior:**
"Failed to download manifest" error after choosing the port and hitting connect.

bmorcelli, 2024-11-21:
Hey buddy, we just changed our main workflow, and it need a new release to work ;)

We are releasing 1.7.2 in a few minutes.. beta version flasher is supposed to be working

bmorcelli, 2024-11-21:
Working now! Thank you!! 🙏🙏

closed 2024-11-21 by bmorcelli

ATOMNFT, 2024-11-21:
> Working now! Thank you!! 🙏🙏

Thank you for your dedication and work on Bruce!!!!