BruceDevices/firmware #462: CYD support (2.4+Capactive)

Teapot321, 2024-11-21:
https://esp.huhn.me/

0х1000 - Bootloader
0х8000 - Partition
0x10000 - Firmware
[image]
[image]

[firmware.zip](https://github.com/user-attachments/files/17843305/firmware.zip)

Asmodeux2, 2024-11-21:
Уже добавили под другим названием CYD-2432W328C_2

Asmodeux2, 2024-11-21:
Я ранее уже просил @bmorcelli добавить его как в WebFlasher так и в Release. Просто название от прошивки донора так и осталось с поддержкой CST820. Которую кстати ты просил

bmorcelli, 2024-11-21:
I need the interface files to add to the project.. 

interface.cpp. interface.h and platformio.ini for this device

Webflasher is driven by an automated workflow, so it won't be possible to only add the binaries there

bmorcelli, 2024-11-21:
Dnoe

closed 2024-11-21 by bmorcelli

ciccirix, 2024-11-21:
[image]
  not work on me..

ciccirix, 2024-11-21:
si ferma su quella schermata

Asmodeux2, 2024-11-21:
> [image]
>   not work on me..

https://telegra.ph/Kak-opredelit-CYD-11-21

You use ESP32-2432S024(R) firmware work on ESP32-2432S024(C)

Asmodeux2, 2024-11-21:
> [image]
>   not work on me..

Use CYD-2432S028 firmware on your ESP32-2432S024R