BruceDevices/firmware #464: Backup Tool / .ZIP unzipper / Costom WiFi Spammer AP names

Int0Y0urM0ther, 2024-11-21:
**Is your feature request related to a problem? Please describe.**
Idk why but anytime i Upgrade the fw, all Files Just Get removed.

**Describe the solution you'd like**
Like a Backup Tool or Something in LittleFS to make a complete Backup of all the Files and settings in the LittleFS Or unzip .ZIP Files in LittleFS.

**Describe alternatives you've considered**
Backup App/Tool
unzip .zip in LittleFS 

Other Feature requests
—————————————
A txt file that contains WiFi names to use in WiFi spammer Like.
WiFi 1
WiFi 2
WiFi 3

**Additional context**
I dont know how it can be done but it would be cool

bmorcelli, 2024-12-07:
Webflasher will always erase the flash..
there are no way to back it up without SD card, so you need to do the following when you want  to update to the last Beta Release
* Download the binary (WebPage brench of this project)
* make sure you have python and pip installed in your computer
* install esptool: `pip install esptool`
* find out what is the COM port your device is connected
* enter the folder you have the binary saved, and run this command:
* `esptool --port COMx write_flash 0x0 Bruce-yourversion.bin `
* please change the "x" for your COM number and the "Bruce-yourversion.bin" with your file name.. 

it will install the binary and won't erase the LittleFS files

closed 2024-12-07 by bmorcelli

Int0Y0urM0ther, 2024-12-12:
Okay ty 👍🏻