BruceDevices/firmware #465: esp32-2432S028 CYD PANTALLA FONDO BLANCO  COLORES MAL

jhonpohac, 2024-11-21:
Hola ayuda desde la 1.7 a 1.72 los colores están mal ayuda por favor 

[image]
[image]
[image]
[image]
**Describe the bug:**

**Steps To Reproduce:**

- Device used: 

**Expected behavior:**

**Actual behavior:**

**Additional context / logs:**

bmorcelli, 2024-11-21:
Install CYD-2USB version.. the only difference is the colors inverted

closed 2024-11-21 by bmorcelli

bmorcelli, 2024-11-21:
Si, instalando la version correcta.. y las dos versiones estan disponibles... Si la esp32-2432S028R no esta buena, instale la CYD-2USB, que la unica difer ncia es la inversion de colores

O se instalaste la version 2USB y los colores estan invertidos, instala la otra, simples

jhonpohac, 2024-11-21:
> Si, instalando la versión correcta.. y las dos versiones están disponibles... Si la esp32-2432S028R no esta buena, instale la CYD-2USB, que la única diferencia es la inversión de colores
> 
> O se instala la versión 2USB y los colores están invertidos, instala la otra, simples

Hola amigo todas las versiones  nuevas que instalo e salen a si

jhonpohac, 2024-11-21:
> > Si, instalando la versión correcta.. y las dos versiones están disponibles... Si la esp32-2432S028R no esta buena, instale la CYD-2USB, que la única diferencia es la inversión de colores.
> > O se instala la versión 2USB y los colores están invertidos, instala la otra, simples
> 
> Hola amigo todas las versiones nuevas que instalan y salen a si

[image]

jhonpohac, 2024-11-21:
esta versión  CYD-2432W328C(inv_colors)_and_CYD-2432S024C que corrugue eso la puedo instalar es que es para otro modelo?

bmorcelli, 2024-11-21:
> esta versión CYD-2432W328C(inv_colors)_and_CYD-2432S024C que corrugue eso la puedo instalar es que es para otro modelo?

Estas versiones son para la pantalla capacitiva

Cual es la version que estás instalando? 

Yo le dije para cambiar de Beta para Last release, es para cambiar de dispositivo, Instala la version CYD-2USB..

bmorcelli, 2024-11-21:
Install this one
[image]

This is what I am talking about

jhonpohac, 2024-11-21:
> Instalar este ![Captura de pantalla_20241121-182138](https://private-user-images.githubusercontent.com/104320209/388699074-4e4174b9-138d-4a49-9c41-7fbe07024109.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzIyMjc5NDIsIm5iZiI6MTczMjIyNzY0MiwicGF0aCI6Ii8xMDQzMjAyMDkvMzg4Njk5MDc0LTRlNDE3NGI5LTEzOGQtNGE0OS05YzQxLTdmYmUwNzAyNDEwOS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMTIxJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTEyMVQyMjIwNDJaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1jY2RmNmM5M2MyODQ2ZWEzYWM2MmFjMTAyOWM0YzdmMTdmMWJlYjUxYjY4NTQ0ZWFmZjFkODNjZDc5Mzk2OGYyJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.nj1rQgfhUEPi2NyF7YnKtglW2IgafSzRKUyBvOwxlzM)
> 
> De esto es de lo que estoy hablando
esa la que tengo instalada pero se ve  a si es la ultima la 1.7.2 esa misma fue la instale

bmorcelli, 2024-11-21:
Instala este ahora

[image]

jhonpohac, 2024-11-21:
> Instala este ahora
> 
> ![Screenshot_20241121-193146.png](https://private-user-images.githubusercontent.com/104320209/388720915-456549bc-cfd1-4cd6-be65-ba6672f00e04.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzIyMzAyNzgsIm5iZiI6MTczMjIyOTk3OCwicGF0aCI6Ii8xMDQzMjAyMDkvMzg4NzIwOTE1LTQ1NjU0OWJjLWNmZDEtNGNkNi1iZTY1LWJhNjY3MmYwMGUwNC5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQxMTIxJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MTEyMVQyMjU5MzhaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT05MzlmNDUyMmM4ZmNmYmMyZDJmOGZiODM1NTU1YTNhODQyMTM4YjZjZWMxODk3ZTcxODY1MWEwNWEyM2Y0ZTY1JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.cR5u-RvY5Hamdn6kWqjkXVywths7sflXu2xFdFr5rzg)

amigo eres el mejor super quedo full  y en la ultima versión ahora a disfrutar quedo super muchas muchas gracias  dare mi aporte al proyecto muchas muchas gracias imagen perfecta  muchas gracias y disculpa las molestias quedo super super

jhonpohac, 2024-11-26:
Hello, first of all, congratulate you for such an excellent job and I would like to ask you that it would be good to have a battery charging module since we have many problems in the m5stack cardputer and in the m5stickc plus 2 to charge it would be good to place the indicator LED red discharged orange charging green full load and I have an Ultrasonic Repellent to repel brave dogs and Anti-Mosquitoes I can pass them on to you the sources add them it would be very good to have them thank you very much for your time attention and please continue this great project and continue improving to become the best and most complete of all

hola primero que todo felicitarlos por tan excelente trabajo y me gustaría pedirle que sería bueno  tener un módulo de carga batería ya que tenemos muchos problemas en el m5stack cardputer y  en el m5stickc plus 2 para cargar sería bueno colocar el led indicador rojo descargada naranja cargando verde full carga  y yo tengo un  Repelente Ultrasonido para ahuyentador de perros bravos y Antimosquitos se los puedo pasar los fuentes  agregarlos sería muy bueno tenerlos muchas gracias por su tiempo atención y que sigan este gran proyecto y sigan mejorado para llegar hacer el mejor y el más completo de todos