BruceDevices/firmware #474: Add iButton feature

geo-tp, 2024-11-23:
It would be nice if Bruce could implement the iButton feature

here is a basic example of read/write for M5StickCPLUS2 (also possible with Cardputer)

#### M5StickCPLUS2 iButton example
```
lib_deps = 
    m5stack/M5Unified@^0.1.17
    paulstoffregen/OneWire@^2.3.8
```

```
#include <OneWire.h>
#include <M5Unified.h> 

#define ONE_WIRE_BUS 0

OneWire oneWire(ONE_WIRE_BUS);
byte buffer[8];

void setup() {
  M5.begin();
  M5.Lcd.setRotation(3);
  M5.Lcd.fillScreen(TFT_BLACK);
  M5.Lcd.setTextColor(TFT_WHITE);
  M5.Lcd.setTextSize(2);
  M5.Lcd.setCursor(8,60);
  M5.Lcd.println("Waiting for iButton");
  delay(100);
}

void write_byte_rw1990(byte data) {
  for (int data_bit = 0; data_bit < 8; data_bit++) {
    delay(25);
    oneWire.write_bit(~data);
    data >>= 1;
  }
}

void write_ibutton() {
  M5.Lcd.setCursor(52,102);
  M5.Lcd.print("Wait...");

  oneWire.reset(); // Reset bus
  oneWire.skip(); // Skip rom check
  delay(20);

  // Step 1 : Prepare for writing
  oneWire.write(0xD1);  // Start write command
  oneWire.reset(); // Reset bus

  // Step 2 : Write the ID
  oneWire.write(0xD5);  // Write ID command
  for (byte i = 0; i < 8; i++) {
    write_byte_rw1990(buffer[i]);  // Write each byte
  }
  oneWire.reset();  // Reset bus

  // Step 3 : Finalise
  oneWire.write(0xD1);  // End of write command
  oneWire.reset();  // Reset bus

  // Display end of copy
  M5.Lcd.clear();
  M5.Lcd.setCursor(90,50);
  M5.Lcd.setTextSize(2);
  M5.Lcd.print("COPIED");
  M5.Lcd.setCursor(40,80);
  M5.Lcd.print("Release button");

  delay(3000);

  M5.Lcd.clear();
  M5.Lcd.setCursor(10,60);
  M5.Lcd.println("Waiting for iButton");
}

void read_ibutton() {
  oneWire.write(0x33);  // Read ID command
  oneWire.read_bytes(buffer, 8);  // Read ID

  // Display iButton
  M5.Lcd.clear();
  M5.Lcd.setTextSize(2);
  M5.Lcd.setCursor(55,20);
  M5.Lcd.println("iButton ID: ");

  // Dislay ID
  M5.Lcd.setTextSize(1.7);
  M5.Lcd.setCursor(12, 57);
  for (byte i = 0; i < 8; i++) {
    M5.Lcd.print(buffer[i], HEX);
    M5.Lcd.print(" ");
  }

  // Display copy infos
  M5.Lcd.setCursor(55,85);
  M5.Lcd.setTextSize(1.5);
  M5.Lcd.println("Hold OK to copy");
}

void loop() {
  M5.update();

  // iButton is plugged
  if (oneWire.reset() != 0) {
    // Main Button is pressed
    if (M5.BtnA.isHolding()) {
      write_ibutton();
    } else {
      read_ibutton();
    }
  }
  delay(500);
}

```

#### Probe and Tags
[image]

#### Hat with Protoboard
[image]

Ademolght, 2024-11-24:
nobody use ibutton in 2025

AnvXprogs, 2024-11-24:
> nobody use ibutton in 2025

are you sure

0n1me, 2024-11-25:
> никто не будет использовать ibutton в 2025 году

[image]
bro...

geo-tp, 2024-11-25:
> nobody use ibutton in 2025

[image]

Sonys9, 2024-12-22:
yeah we need it

XOPC126, 2025-02-09:
good afternoon. Can you tell me how to connect this to a card computer?

Maksimius, 2025-03-31:
Only read works ((
m5stickplus2
fw v1.9.1 ibutton rw1990

Maksimius, 2025-04-06:
Fix write iButtons.
https://github.com/pr3y/Bruce/pull/1048