BruceDevices/firmware #475: bad usb at cyd

khanh-lol, 2024-11-24:
hi bruce team, how to use badusb at cyd?

Ademolght, 2024-11-24:
> hi bruce team, how to use badusb at cyd?

imposible

BeqaMiqa, 2024-11-24:
here is the guide for it, but there isn't cyd mentioned anywhere:
https://github.com/pr3y/Bruce/wiki/Others#badusb

in short you will have to buy a CH9329 module. If you do, update us please.

khanh-lol, 2024-11-24:
> here is the guide for it, but there isn't cyd mentioned anywhere: https://github.com/pr3y/Bruce/wiki/Others#badusb
> 
> in short you will have to buy a CH9329 module. If you do, update us please.

im have it the CH9329, but what i can do now?

BeqaMiqa, 2024-11-24:
> > here is the guide for it, but there isn't cyd mentioned anywhere: https://github.com/pr3y/Bruce/wiki/Others#badusb
> > 
> > in short you will have to buy a CH9329 module. If you do, update us please.
> 
> im have it the CH9329, but what i can do now?

IMPORTANT:
I am not familiar with the CH9329 module. If the 5v pin is used for externality power other devices the first method should work if it is not designed for that do NOT try the first method 

At the bottom next to the micro USB or type c (depends on which version cyd you have) there are 4 pins labeled VIN RX TX GND.

1.this is how the connections will go:
Cyd    CH9329 module
Vin  ->   Vcc (if and only if this pin is used to external power)
Rx   ->    tx
Tx   ->    rx
Gnd ->   gnd
This should theoretically work, maybe. I don't know 

vvv(TRY THIS FIRST)vvv
2.If the vcc pin is NOT used for powering external devices you can just only do this connection:
Cyd    CH9329 module
Rx   ->    tx
Tx   ->    rx
Let me know if this works

khanh-lol, 2024-11-25:
ty man

BeqaMiqa, 2024-11-25:
If you don't mind could you comment few pictures or a short video of it working please

khanh-lol, 2024-11-27:
> If you don't mind could you comment few pictures or a short video of it working please

my computer dont have a camera, sorry

khanh-lol, 2024-11-27:
it working

khanh-lol, 2024-11-27:
burhh someone is stole my CH9329

khanh-lol, 2024-11-27:
3$ :\

BeqaMiqa, 2024-11-27:
Also I forgot if ch9329 module outputs 5v on it's TX pin you'll need a level shifter so it won't fry the esp

khanh-lol, 2024-11-28:
> Also I forgot to say, if ch9329 module outputs 5v on it's TX pin you'll need a level shifter so it won't fry the esp

k

closed 2024-12-07 by bmorcelli