BruceDevices/firmware #477: Question

VladGamer43, 2024-11-24:
Hi want to ask, can I use a PN532 module on my cardputer and are the features same as on the M5Stack rfid2 module?

VladGamer43, 2024-11-24:
And can I emulate a tag with it , because I read that this module can emulate tags

rennancockles, 2024-11-24:
Yes, you can use a PN532 on your cardputer. At the moment you'll have same functionality as you have with M5stack's RFID2. Although PN532 has some emulation capabilities, you cannot fully emulate a tag with it.

VladGamer43, 2024-11-24:
Okay

VladGamer43, 2024-11-24:
And One question I want to buy 1 more M5Stack cardputer but I live in Germany where can I buy it the cheapest with shipping

rennancockles, 2024-11-24:
I don't know what would be the best place for you, but for us here in Brazil it's AliExpress. You can also check m5stack's official website

closed 2024-11-25 by rennancockles