BruceDevices/firmware #485: Request for Smaller Firmware Version (Bruce-CYD-2USB.bin)

Minhhoang-dpetrai, 2024-11-26:
Hi

I am using the Bruce firmware on my ESP32, but I encountered an issue with the file size of Bruce-CYD-2USB.bin. The current size (3.4MB) exceeds the program storage limit of my ESP32 board (1.3MB). Unfortunately, I don't have access to a computer to modify or flash a custom partition scheme.

Could you please provide a smaller or optimized version of the firmware that fits within the 1.3MB program space? Alternatively, are there any steps I can follow to reduce the size on my own using just a mobile device?

Thank you for your hard work and support!

tinhnguyenxuan, 2024-11-26:
có phiên bản laucher đấy thi xó 2mb

Minhhoang-dpetrai, 2024-11-26:
Where?

bmorcelli, 2024-11-26:
> Could you please provide a smaller or optimized version of the firmware that fits within the 1.3MB program space? Alternatively, are there any steps I can follow to reduce the size on my own using just a mobile device?

Are you trying to use it whit Launcher? if so, you must [use this version](https://github.com/pr3y/Bruce/blob/WebPage/BetaRelease/Bruce-LAUNCHER_CYD-2USB.bin), otherwise it wont fit Launcher APP partition.

CYDs have 4Mb of flash memory, and it has 1,5Mb of LittleFS storage that is supposed to be used to save simple data (html, sub, ir files) saving the binary file there won't change anything.

Launcher can be [installed here](https://bmorcelli.github.io/M5Stick-Launcher/) but you'll need a computer, anyways

Minhhoang-dpetrai, 2024-11-26:
Sketch too big: uses 2455776 of program storage space. Maximum is 1310720 bytes. Helpp

bmorcelli, 2024-11-26:
Where are you trying to install?

Board need to have at least 4mb of storage, otherwise it won't fit..

Give us more information, because your story makes no sense

Minhhoang-dpetrai, 2024-11-26:
I am installing it on my phone and I use the esp32 loader app

bmorcelli, 2024-11-26:
Well.. it is not possible then..

To update on the go I advise you to find a PC and install Launcher.. then you will be able to install the CYD Launcher versions on it, and some others applications using Launcher web interface