BruceDevices/firmware #489: T-Embed CC1101 battery issues

SteveHawk, 2024-11-28:
**Describe the bug:**

I flashed bruce 1.7.2 onto my Lilygo T-Embed CC1101, and observed two battery issues:

1. Battery level is not quite stable, often fluctuating up and down for up to 5-10%.
2. Battery drain. A full charge cannot last a day sitting idle, which feels short for its 1300mAh battery. 

Are these issues expected?

**Steps To Reproduce:**

- Device used: T-Embed CC1101

**Expected behavior:**

**Actual behavior:**

**Additional context / logs:**

hahmann, 2024-12-06:
I can confirm the same behavior with my T-Embed CC1101 on cases (1) and (2).
Also, to note, let's say the battery is at 38%, if I connect it to the USB-C it goes up to 70%.
If right after I disconnect the USB-C it goes back to the 38% it was before.

Any additional tests needed, please let me know.

bmorcelli, 2024-12-07:
Battery values are mesured by bus voltage, so when you plug the cable the voltage goes up and battery starts charging..  this device doesn't have a method to read the battery only, so this behavior is expected.

> Battery drain. A full charge cannot last a day sitting idle, which feels short for its 1300mAh battery.

that is why you can put ir to sleep and have a more time to use..
While in main menu, keep the upper btn pressed and roll the wheel counter clockwise, it will start a count down and will turn it off (put into deepsleep), saving battery

closed 2024-12-07 by bmorcelli