BruceDevices/firmware #491: Interface optimization

Teapot321, 2024-11-28:
Hello, I want to ask you to optimize the interface. There are extra sections that need to be removed to the Others tab: Scripts (it definitely shouldn't be on the main page), Clock (a function that may be useless on some devices, for example CYD, after each shutdown, time data is erased), Connect (a function that is quite useful, but not in demand due to bugs).
And now about the sections that should be added to the main page: Bad KB(must be added to the main page, and must contain BadUSB and BadBLE).

Please do it😀

IamNobody96, 2024-11-28:
Hi.
I'm not a contributor but I could make that.

9dl, 2024-11-29:
I'm working on this :)

eadmaster, 2024-11-30:
related: https://github.com/pr3y/Bruce/issues/326

closed 2024-11-30 by pr3y