BruceDevices/firmware #494: Adapt for DIY assembled device on ESP32 board

dgorbatko, 2024-11-28:
Hi, firtst of all, big thanks for the amazing rom! You did an incredible work.

I played a lot with your rom on my M5Stickc Plus2 but i wanted to create my own custom device and adapt your rom for this device. Bruce rom on M5Stick is greate but the screen is too small for my eyes and I wanted to use dual nrf and have all modules connected together using the same MISO MOSI SCL with different CS and CE pins.
Devices that I have:
1) Main board ESP32-S3 Wroom-1 N16R8 (with psram and 16mb flash) with 27 free GPIOs (there are more, but i believe i can't use them for the modules)
2) TFT SPI screen 320x240 ST7789v (I could made it works only with Adafruit GFX + adafruit st7789 libraries, but couldn't with TFT_eSPI, but I believe the issue is in the correct driver or settings, so its solveble) (GND, VCC, SCL, SDA, RST, DC, CS, BL)
3) 5 direction joystick, to more easy to use the onscreen keyboard (COM, UP, DWN, LFT, RHT, MID, SET, RST)
4) 1 simple button to use it as "Back" or "Cancel" button.
5) 2 NRF24 (GND, 3V, CE, CSN, SCK, MOSI, MISO)
6) CC1101 (GND, 3V, GDO0, CSN, SCK, MOSI, MISO)
7) Micro SD card reader. ( CS, SCK, MOSI, MISO, VCC, GND)
8) IR reciever transmitter (that worker with M5Stick without any problem).
9) PN532 (GND, VCC, SDA, SCL)
10) FM SI4713: RST, SCL, SDA, GND, VCC
11) Other things like 1S Lipo charger, 3.7 to 3.3 step down, lipo battery, capasitors for NRF modules, resistors etc..

I've created a pinout table:
[image]

I spent two weeks exploring and editing your source code but no luck at all. Or it simply wont build or it builds, uploads, but device goes to bootloop or it boots but (based on serial log) with a lot of errors about files system mounting errors etc, and screen doesn't work.
The Idea is to at least adapt your rom for the custom hardware with hardcoded pins (without option to select pins for the modules) to let everything works even without the idea of dual nrf. 

Maybe you have some ideas / tutorials or simplifyed source code how to make it?

Big thanks in advance!

eadmaster, 2024-11-30:
> or it builds, uploads, but device goes to bootloop or it boots but (based on serial log) with a lot of errors about files system mounting errors etc

`board_build.partitions` is likely the cause for this, try using the default partition scheme. E.g. just comment this line like i did here:

https://github.com/pr3y/Bruce/blob/main/ports/ESP-General/platformio.ini#L14

Also, try to start from scratch in headless mode, and connect one module at a time, since they may interfere with each other.

dgorbatko, 2024-12-01:
Hi, thanks for the answer!
Yes I did try this. Anf its already commented in the fresh just downloaded rom. So only one thing that I did is to comment all env's in root platformio file and added esp32 env there, like this:

[platformio]
default_envs = esp32-s3-devkitc-1-psram
	;m5stack-cplus1_1
	;m5stack-cplus2
	;m5stack-cardputer
	;m5stack-core2
	;m5stack-core16mb
	;m5stack-core4mb
	;m5stack-cores3
	;CYD-2432S028
	;CYD-2USB
	;lilygo-t-embed-cc1101
	;lilygo-t-deck
	
So I flashed this freshly downloaded rom without any edits (no screen settings, no pins settings) to headless device, without anything connected, just esp32, and the log is the following:

Connected!
{
  "priColor": "a80f",
  "secColor": "880f",
  "bgColor": "0",
  "rot": 1,
  "dimmerSet": 10,
  "bright": 100,
  "tmz": 0,
  "soundEnabled": 1,
  "wifiAtStartup": 0,
  "webUI": {
    "user": "admin",
    "pwd": "bruce"
  },
  "wifiAp": {
    "ssid": "BruceNet",
    "pwd": "brucenet"
  },
  "wifi": {},
  "irTx": 40,
  "irRx": 46,
  "rfTx": 8,
  "rfRx": 46,
  "rfModule": 0,
  "rfFreq": 433.92,
  "rfFxdFreq": 1,
  "rfScanRange": 3,
  "rfidModule": 0,
  "wigleBasicToken": "",
  "devMode": 0
}wifiConnect
IP: 172.0.0.1
startWebUi
IP: 172.0.0.1
Configuring Webserver ...
[  8137][E][esp32-hal-misc.c:143] disableCore1WDT(): Failed to remove Core 1 IDLE task from WDT

I don't think that the last row is ok.

And i see that there is PIN 46 coded to GROVE_SCL in the platformio file in ESP32 folder, but on my esp32-S3 Wroom1 N16R8 pin46 is reserved for some system things [based on the official description](http://wiki.fluidnc.com/en/hardware/ESP32-S3_Pin_Reference): "gpio.46 Boot mode. Weak pull down during reset. (Enabling/Disabling ROM Messages Print During Booting)". I'm not sure but maybe this also could cause some errors.
Also I don't fully understand how display settings should be applied in Bruce rom for ESP32 because it looks like the simple way of setting it up in User_Setup.h (driver, mosi and other pins, size and other things) doesn't work, its still continue to hookup a wrong mosi pin in the code. I found it if i hover the coursor over the TFT_MOSI variable in the code, it references to some another pin.

Another thing that's not clear for me is the spi buses, based on code, it creates two spi buses, one for TFT + SD and second for NRF + CC1101. So if its true, what pins should be coded to TFT and SD MOSI and CLK?

eadmaster, 2024-12-01:
If you are building without the `HAS_SCREEN` flag the board should autostart the webui, try connecting via wifi.

ESP32 boards usually  have 3 set of pins that can be used for SPI, look in the official docs. Some Spi devices can share the same pins (except the CS = chip select), some don't.
 I haven't tried with an lcd yet, but they may require a separate bus to get a good framerate.

> gpio.46 Boot mode. Weak pull down during reset. (Enabling/Disabling ROM Messages Print During Booting)". I'm not sure but maybe this also could cause some errors.

Not all esp32 boards have the same pinouts, you may need to change some pins in the platformio.ini and recompile.

dgorbatko, 2024-12-01:
Your right, i have no HAS_SCREEN = 1 in my platformio file and I really can connect to the webui.
Btw there it shows I have only 1.5MB total, but based on my board N16R8 I need to have 16MB. Probably i need to figure out how to use partition table.

dgorbatko, 2024-12-04:
Hi, do you have any recomendation which module should sits on which spi bus (with different CS/DC etc...),
Like I have:
TFT - MOSI SCL (works only on HSPI BUS)
NRF24 1 - MOSI MISO SC
NRF24 2 - MOSI MISO SC
CC1101 - MOSI MISO SC
SD Reader - MOSI MISO SC

We can ignore the second NRF24, lets figure out the connection only with one NRF24 for now.
Currently I succesfully booted my diy device. Setup 6 pins joystick to one pin with different resistors, and adjusted the navigation like up down left right ok cancel. Also succesfuly checked one NRF24 that sits on different pins than TFT Display.
But now im stuck on logic where to connect SD, CC1101 and second NRF to make it work.
Any advices?
Thanks!

closed 2024-12-07 by bmorcelli

khaleeek, 2025-05-06:
Hi could you share code or help I have same esp32 s3 44 pins 2 type c
A st77 2.8 inch touch display.