BruceDevices/firmware #498: Command and Control

Jethro-San, 2024-11-29:
This sounds a little ridiculous even to me, but would it be possible to implement a Mirai type botnet command and control feature, or some sort of command-and-control server with a bad USB script to enter the device into the server via the command prompt and include some features such as remote Rickrolling and access to command prompt on the system.

Jethro-San, 2024-12-02:
not necessarily, something like https://github.com/offsecginger/koadic where somehow something has to be entered into the command prompt on the system to connect to the server, sometimes it works with SSH, but often it works by running some bash on the target machine

9dl, 2024-12-02:
Oh okey, I can look into it.