BruceDevices/firmware #502: Kindly ask for the implementation of the support for GPS v1.1 module

Taranovski, 2024-12-01:
Bought GPS v1.1 module, thought that it should be better than the "regular" GPS module - it looks like it is not supported :(

bmorcelli, 2024-12-07:
available on BETA

closed 2024-12-07 by bmorcelli

Taranovski, 2024-12-08:
checked the update - can select the baudrate, but then it still seems like it doesn't work as expected.
for gps tracker, I have:

getting info on the screen (example):
initially:
gps location not updated
date: 2013-58-207
time: 46:51:37
sat: 0
hddp: 25.50

after some time:
gps location not updated
date: 2024-12-08
time: 22:06:12 <--- seems like UTC? no timezone? should be EST at some point?
sat: 0
hddp: 25.50

I tried to reach a maintainer of another repo, which provides gps tracking for cardputer:
https://github.com/geo-tp/M5Cardputer-GPS-Logger/issues/1

with that bin file (mentioned in the discussion) - I get the proper current time updates, no lat and long though.
That maintainer says that, apparently, for gps module v1.1 (at6668 + max2659) some other library should be used (compared to the standard gps module) - I planned to look it up myself, but, unfortunately, don't have time to do it in the next few weeks.

I'm also unsure about how to validate if my module is actually working - so far it is suspicious that 2 apps yield similar results, it might be the case that my module is broken somehow.