BruceDevices/firmware #509: When will this feature be added? (PN532 Emulation)

Teapot321, 2024-12-02:
Why haven't we added the pn532 emulation yet, you've already shown its demonstration and as I understand it, it's already been done

Video from the developer
https://github.com/user-attachments/assets/313906bc-66ce-4e4d-a66d-0154c2370d6f

closed 2024-12-02 by bmorcelli

jakre1234, 2024-12-16:
Can the emulation be done with the m5 rfid module?

> Because PN532 changes the 3rd bit of the emulated tag, invalidating the whole thing..
> 
> So the tag you are emulating is different from the one you are trying to emulate, and it won't work at all..

can the emulation be done with m5stack rfid module?

bmorcelli, 2024-12-16:
No, it is not possible

jakre1234, 2024-12-16:
So I can not simply emulate now or in future but can write the data to any rfid card which supports rewriting block 0 and use that card, right?

> No, it is not possible
So I can not simply emulate now or in future but can write the data to any rfid card which supports rewriting block 0 and use that card, right?

So I can not simply emulate now or in future but can write the data to any rfid card which supports rewriting block 0 and use that card, right?

bmorcelli, 2024-12-16:
It can read and write tags (I can't tell you about block 0, I personally don't know).

Emulation is not available on M5Stack RFID/2 modules, only in PN532 with some restrictions

CloudWaddie, 2025-05-05:
https://github.com/Seeed-Studio/PN532/issues/88